AlexandreRouma/SDRPlusPlus #1: Can't find volk and soapysdr

LukeRPi, 2020-07-20:
Can I have a link to download volk and soapysdr dll? I can't find them.

AlexandreRouma, 2020-07-20:
The easiest way to get them on windows is to install PothosSDR.
You can download the latest version here 
https://downloads.myriadrf.org/builds/PothosSDR/

starvald, 2020-07-20:
i tried copying volk and soapysdr dll's from the Pothos folder and it wouldn't run, so copied all the files from sdr++  into C:\Program Files\PothosSDR\bin and it worked there.

LukeRPi, 2020-07-20:
> i tried copying volk and soapysdr dll's from the Pothos folder to the unpacked sdr++ one and it wouldn't run, so copied all the files from sdr++ into C:\Program Files\PothosSDR\bin and it worked there.

Ok, thanks. I've done it and it starts but when I press the play button it doesn't show the waterfall.

LukeRPi, 2020-07-20:
`[INFO] [UHD] Win32; Microsoft Visual C++ version 14.0; Boost_106700; UHD_3.15.0.0-0-gaea0e2de
Found Rafael Micro R820T tuner
[INFO] Opening Generic RTL2832U OEM :: 00000001...
Found Rafael Micro R820T tuner
Starting DSP Thread!
32000 800
FloatResamp: 800 192Success
Success
Loading icons...
Exact sample rate is: 250000.000414 Hz
rtlsdr_demod_write_reg failed with -9
r82xx_write: i2c wr failed=-9 reg=0a len=1
Setting sample rate to 250000.000000
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
DEBUG: 1288
FIRResampler.setInputSampleRate(): 4 5
1250 1000
Stopping FM demodulator
Starting FM demodulator
DEBUG: 168
FIRResampler.setOutputSampleRate(): 4 5
1250 1000
FloatFIRResampler.setInputSampleRate(): 6 25
FloatResamp: 1000 240Pos changed
Resized: 844 354
rtlsdr_demod_write_reg failed with -9
r82xx_write: i2c wr failed=-9 reg=17 len=1
r82xx_set_freq: failed=-9
[INFO] Using format CF32.
`

This is the log from the command prompt.

AlexandreRouma, 2020-07-20:
> i tried copying volk and soapysdr dll's from the Pothos folder to the unpacked sdr++ one and it wouldn't run, so copied all the > files from sdr++ into C:\Program Files\PothosSDR\bin and it worked there.

This means your path variable isn't set up properly.
This is likely because you didn't select "add to path" (or something similar) in the pothos installer.

To fix this, type "Environment Variable" in the start menu, then click on "Edit the system environment variables".  Click on "Environment Variables" and, in the new window that shows up, double click on "Path" in the "System Variables" list. A new window will pop up. You can now click on "new" and paste "C:/Program Files/PothosSDR/lib/"
click again on new and paste "C:/Program Files/PothosSDR/bin". Now click "ok", then "ok" and finally "apply".
Restart your computer and the libs will have been added to your path.

you SHOULDN'T paste all files in the pothos dir, if you did than, remove them (make sure they're the one from the zip file).

> `[INFO] [UHD] Win32; Microsoft Visual C++ version 14.0; Boost_106700; UHD_3.15.0.0-0-gaea0e2de Found Rafael Micro R820T tuner [INFO] Opening Generic RTL2832U OEM :: 00000001... Found Rafael Micro [...]

Try running it at a higher sample rate. From what I can see in the logs, you're running at 250ks/s, this is too low
Select a higher sample rate in the menu under the device list. I suggest using 2560000 since you're using an RTL-SDR.

closed 2020-07-20 by AlexandreRouma