AlexandreRouma/SDRPlusPlus #3: It starts, but doesn't see RTL-SDR

Radiomix2000, 2020-07-21:
I don't understand why it says me to upgrade libusb. I just installed fresh zadig.
Any help?

`[INFO] [UHD] Win32; Microsoft Visual C++ version 14.0; Boost_106700; UHD_3.15.0.0-0-gaea0e2de
←[1m←[31m[ERROR] rtlsdr_get_device_usb_strings(0) failed←[0m
WARNING: Skipping broken USB device. Please upgrade libusb.
WARNING: Skipping broken USB device. Please upgrade libusb.
Starting DSP Thread!
Resamp: 32000 800FloatResamp: 800 192Success
Success
Loading icons...
Setting sample rate to 4000.000000
FIRResampler.setInputSampleRate(): 50 1
Resamp: 20 1000Stopping FM demodulator
Starting FM demodulator
FIRResampler.setOutputSampleRate(): 50 1
Resamp: 20 1000FloatFIRResampler.setInputSampleRate(): 6 25
FloatResamp: 1000 240Pos changed
Resized: 844 354`

AlexandreRouma, 2020-07-21:
Hi,
Have you tried using this RTL-SDR in another software?
I see you're running windows, you should try to verify if it works with something like SDR#

If it does work in SDR# but not SDR++, run these commands and post the result:
```
SoapySDRUtil --info
```
and
```
SoapySDRUtil --list
```

Cheers

Radiomix2000, 2020-07-21:
Yes it works with HDSDR, SDR# and SDRangel

Radiomix2000, 2020-07-21:
C:\PothosSDR\bin>SoapySDRUtil.exe --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-PothosSDR-2020.01.26-vc14-x64
API Version: v0.8.0
ABI Version: v0.8
Install root: C:\PothosSDR
Search path:  C:\PothosSDR/lib/SoapySDR/modules0.8
Module found: C:\PothosSDR/lib/SoapySDR/modules0.8/airspyhfSupport.dll (0.1.1-81ca737)
Module found: C:\PothosSDR/lib/SoapySDR/modules0.8/airspySupport.dll   (0.1.2-10d697b)
Module found: C:\PothosSDR/lib/SoapySDR/modules0.8/audioSupport.dll    (0.1.1-2ae84e3)
Module found: C:\PothosSDR/lib/SoapySDR/modules0.8/bladeRFSupport.dll  (0.4.1-1c1e8aa)
Module found: C:\PothosSDR/lib/SoapySDR/modules0.8/HackRFSupport.dll   (0.3.3-3c514ce)
Module found: C:\PothosSDR/lib/SoapySDR/modules0.8/IrisSupport.dll     (2019.07.0.1-423a351)
Module found: C:\PothosSDR/lib/SoapySDR/modules0.8/LMS7Support.dll     (19.04.1-538e186b)
Module found: C:\PothosSDR/lib/SoapySDR/modules0.8/miriSupport.dll     (0.2.6-5a9674c)
Module found: C:\PothosSDR/lib/SoapySDR/modules0.8/netSDRSupport.dll   (0.1.0-dc4e579)
Module found: C:\PothosSDR/lib/SoapySDR/modules0.8/osmosdrSupport.dll  (0.2.6-5a9674c)
Module found: C:\PothosSDR/lib/SoapySDR/modules0.8/PlutoSDRSupport.dll (0.2.0-e28e4f5)
Module found: C:\PothosSDR/lib/SoapySDR/modules0.8/RedPitaya.dll       (0.1.1-3d576f8)
Module found: C:\PothosSDR/lib/SoapySDR/modules0.8/remoteSupport.dll   (0.5.2-6d9bd82)
Module found: C:\PothosSDR/lib/SoapySDR/modules0.8/rtlsdrSupport.dll   (0.3.1-8ba18f1)
Module found: C:\PothosSDR/lib/SoapySDR/modules0.8/sdrPlaySupport.dll  (0.3.0-14ec39e)
Module found: C:\PothosSDR/lib/SoapySDR/modules0.8/uhdSupport.dll      (0.3.6-47972ba)
Available factories... airspy, airspyhf, audio, bladerf, hackrf, iris, lime, miri, netsdr, osmosdr, plutosdr, redpitaya, remote, rtlsdr, sdrplay, uhd
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

Radiomix2000, 2020-07-21:
C:\PothosSDR\bin>SoapySDRUtil.exe --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[INFO] [UHD] Win32; Microsoft Visual C++ version 14.0; Boost_106700; UHD_3.15.0.0-0-gaea0e2de
Found Rafael Micro R820T tuner
Found device 0
  default_input = True
  default_output = True
  device_id = 0
  driver = audio
  label = Default Device

Found device 1
  default_input = False
  default_output = False
  device_id = 9
  driver = audio
  label = Microphone (Yeti Stereo Microphone)

Found device 2
  default_input = False
  default_output = False
  device_id = 10
  driver = audio
  label = VoiceMeeter VAIO3 Output (VB-Audio VoiceMeeter VAIO3)

Found device 3
  default_input = False
  default_output = False
  device_id = 11
  driver = audio
  label = Microphone (Realtek USB Audio)

Found device 4
  default_input = False
  default_output = False
  device_id = 12
  driver = audio
  label = VoiceMeeter Output (VB-Audio VoiceMeeter VAIO)

Found device 5
  default_input = False
  default_output = False
  device_id = 13
  driver = audio
  label = Microphone (Sonic Studio Virtual Mixer)

Found device 6
  default_input = False
  default_output = False
  device_id = 14
  driver = audio
  label = VoiceMeeter Aux Output (VB-Audio VoiceMeeter AUX VAIO)

Found device 7
  device = plutosdr
  driver = plutosdr
  uri = usb:5.6.5

Found device 8
  device = plutosdr
  driver = plutosdr
  uri = usb:5.5.5

Found device 9
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 00000001
  manufacturer = Realtek
  product = RTL2838UHIDIR
  serial = 00000001
  tuner = Rafael Micro R820T

AlexandreRouma, 2020-07-21:
Thanks for the logs,
This seems be a bug in SoapyRTLSDR.
Since it's just a warning, you can safely ignore it, the SDR should still work just fine.

AlexandreRouma, 2020-07-21:
Forgot about the part where the SDR doesn't up, sorry 😅
I can't reproduce the issue on any system and the logs from SoapySDRUtil seem to suggest that it's still finding the SDR. (weird since it's not showing up in SDR++'s menu)
I'll try to look deeper into the issue once I get the time.

Radiomix2000, 2020-07-22:
I don't know am I right, buy it seems, that all libs are in the right place, but the app trying to find some dll in wrong place. Some paths somewhere may be wrong. But it's just hypothesis, cause there is no info in any logs.
Is there any type of -verbose option to start the program? Or debug mode?

closed 2020-10-22 by AlexandreRouma