AlexandreRouma/SDRPlusPlus #4: Building on OSX instruction & add bookmarks

ah1102, 2020-07-23:
Please add instructions for building your app for OSX. I can participate and collect on my computer under your guidance.

I would also like to see tabs for frequencies. Something similar to the implementation in sdr # or gqrx with spectrum highlighting.

dmaltsiniotis, 2020-07-23:
I'm pretty close to having this working, just need to install the dependent libs with homebrew and tweak the cmake file a bit. I'm just wrestling with OpenGL right now, but if I can get the app to compile cleanly, I'll post my results.

AlexandreRouma, 2020-07-24:
Hi,
A lot of changes need to be made to the display code in order to support MacOS.
Apple being apple, they decide to drop OpenGL support so the renderer needs to be changed to run with Metal.

As for the frequency table, it's a planned, though I'm not sure if I'll implement it before or after multi-vfo and modules.

dmaltsiniotis, 2020-07-24:
I see @AlexandreRouma,

After I resolved the dependencies and cmake file, I ran into segfaults with OpenGL, this is what I'm guessing you were alluding to.

`build % ./sdrpp 
AddressSanitizer:DEADLYSIGNAL
=================================================================
==15254==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000318 (pc 0x7fff38bc6980 bp 0x7ffee13d9510 sp 0x7ffee13d9510 T0)
==15254==The signal is caused by a READ memory access.
==15254==Hint: address points to the zero page.
    #0 0x7fff38bc697f in glGenTextures (libGL.dylib:x86_64+0x197f)
    #1 0x10e914a2f in ImGui::WaterFall::WaterFall() waterfall.cpp:101
    #2 0x10e9013e3 in _GLOBAL__sub_I_main_window.cpp main_window.cpp
    #3 0x11c7051e2 in ImageLoaderMachO::doModInitFunctions(ImageLoader::LinkContext const&) (dyld:x86_64+0x1b1e2)
    #4 0x11c7055ed in ImageLoaderMachO::doInitialization(ImageLoader::LinkContext const&) (dyld:x86_64+0x1b5ed)
    #5 0x11c70000a in ImageLoader::recursiveInitialization(ImageLoader::LinkContext const&, unsigned int, char const*, ImageLoader::InitializerTimingList&, ImageLoader::UninitedUpwards&) (dyld:x86_64+0x1600a)
    #6 0x11c6fe013 in ImageLoader::processInitializers(ImageLoader::LinkContext const&, unsigned int, ImageLoader::InitializerTimingList&, ImageLoader::UninitedUpwards&) (dyld:x86_64+0x14013)
    #7 0x11c6fe0b3 in ImageLoader::runInitializers(ImageLoader::LinkContext const&, ImageLoader::InitializerTimingList&) (dyld:x86_64+0x140b3)
    #8 0x11c6ec5e5 in dyld::initializeMainExecutable() (dyld:x86_64+0x25e5)
    #9 0x11c6f1af7 in dyld::_main(macho_header const*, unsigned long, int, char const**, char const**, char const**, unsigned long*) (dyld:x86_64+0x7af7)
    #10 0x11c6eb226 in dyldbootstrap::start(dyld3::MachOLoaded const*, int, char const**, dyld3::MachOLoaded const*, unsigned long*) (dyld:x86_64+0x1226)
    #11 0x11c6eb024 in _dyld_start (dyld:x86_64+0x1024)

==15254==Register values:
rax = 0x0000000000000000  rbx = 0x000000010eb4f9c0  rcx = 0x0000100021de407d  rdx = 0x000000010ef20328  
rdi = 0x0000000000000001  rsi = 0x0000000000000001  rbp = 0x00007ffee13d9510  rsp = 0x00007ffee13d9510  
 r8 = 0x000000000000005f   r9 = 0x000000008a46ee37  r10 = 0x000000011c782420  r11 = 0x00007fff38bc696e  
r12 = 0x000000010eb4f9c0  r13 = 0x0000100000000000  r14 = 0x000000010ef20330  r15 = 0x000000010ef203c8  
AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (libGL.dylib:x86_64+0x197f) in glGenTextures
==15254==ABORTING
zsh: abort      ./sdrpp
`

I'll wait until if/when there's support for macOS I suppose. I'm not versed enough with OpenGL and metal to tackle a refactoring to make this work on macOS (and not mangle your code in the process.)

Thanks,

-DM

AlexandreRouma, 2020-10-22:
SDR++ now builds and runs on OSX (experimental branch)

closed 2020-10-22 by AlexandreRouma

howard0su, 2020-10-22:
The bug which caused crash is fixed.