AlexandreRouma/SDRPlusPlus #10: linux: sdrplay failure.

starvald, 2020-08-10:
have an airspy R2 which works ok but not with an sdrplay rsp2.

```
[lee@Plasma sdrpp]$ ./sdrpp 
[2020-08-10 20:26:35.886] [info] SDR++ v0.2.5_alpha
ALSA lib pcm_dmix.c:1090:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm_route.c:877:(find_matching_chmap) Found no matching channel map
connect(2) call to /dev/shm/jack-1000/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed
connect(2) call to /dev/shm/jack-1000/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_dmix.c:1090:(snd_pcm_dmix_open) unable to open slave
connect(2) call to /dev/shm/jack-1000/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed
[2020-08-10 20:26:36.092] [info] Audio device open.
[2020-08-10 20:26:36.092] [info] Loading icons
[2020-08-10 20:26:36.107] [info] Loading band plans
[2020-08-10 20:26:36.108] [info] Loading band plans color table
[2020-08-10 20:26:36.108] [info] Loading test module
[2020-08-10 20:26:36.108] [info] Ready.
[2020-08-10 20:26:36.125] [info] Changed input device: 0
[2020-08-10 20:26:36.125] [info] Changed sample rate: 0
=================================================================
==3375754==ERROR: AddressSanitizer: invalid alignment requested in aligned_alloc: 32, alignment must be a power of two and the requested size 0x2710 must be a multiple of alignment (thread T58)
    #0 0x7f41d72bbfc9 in __interceptor_aligned_alloc /build/gcc/src/gcc/libsanitizer/asan/asan_malloc_linux.cpp:192
    #1 0x7f41d6604650 in volk_malloc (/usr/lib/libvolk.so.2.3+0x5c650)

==3375754==HINT: if you don't care about these errors you may set allocator_may_return_null=1
SUMMARY: AddressSanitizer: invalid-aligned-alloc-alignment /build/gcc/src/gcc/libsanitizer/asan/asan_malloc_linux.cpp:192 in __interceptor_aligned_alloc
Thread T58 created by T0 here:
    #0 0x7f41d72611c7 in __interceptor_pthread_create /build/gcc/src/gcc/libsanitizer/asan/asan_interceptors.cpp:214
    #1 0x7f41d6426e49 in __gthread_create /build/gcc/src/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/x86_64-pc-linux-gnu/bits/gthr-default.h:663
    #2 0x7f41d6426e49 in std::thread::_M_start_thread(std::unique_ptr<std::thread::_State, std::default_delete<std::thread::_State> >, void (*)()) /build/gcc/src/gcc/libstdc++-v3/src/c++11/thread.cc:135

==3375754==ABORTING
[lee@Plasma sdrpp]$ 
```

nmaster2042, 2020-08-11:
I have the same behavior with my sdrplay rspduo, when I select sample rate = 125 000 or 250 000.

All other sample rates are ok on my system.

starvald, 2020-08-11:
How to change it though, as soon as I start sdrpp it errors

nmaster2042, 2020-08-11:
I don't know because I can actually start sdrpp.
I select Source panel, then my sdrplay.
The default sample rate checked in the box is 62 500 witch doesn't crash the app.

I think this issue is related to sample rate and the UI.

What is your Soapy version, soapySDRPlay version and SDRPlay API version ?

mine is:

$ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-ga489f3dc
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/libnetSDRSupport.so  (0.1.0-11e598c)
Module found: /usr/local/lib/SoapySDR/modules0.8/libremoteSupport.so  (0.5.2-6d9bd82)
Module found: /usr/local/lib/SoapySDR/modules0.8/librtlsdrSupport.so  (0.3.1-bec4f05)
Module found: /usr/local/lib/SoapySDR/modules0.8/libsdrPlaySupport.so (0.3.0-3381015)
Available factories... netsdr, remote, rtlsdr, sdrplay
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

$ SoapySDRUtil --probe 

######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 
[INFO] devIdx: 0
[INFO] hwVer: 3
[INFO] rspDuoMode: 1
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 0.000000

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=18XXXXXXXX - Single Tuner
  sdrplay_api_api_version=0.000000
  sdrplay_api_hw_version=3

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * RF Gain Select - RF Gain Select
       [key=rfgain_sel, default=4, type=string, options=(0, 1, 2, 3, 4, 5, 6, 7, 8, 9)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]
     * ExtRef Enable - External Reference Control
       [key=extref_ctrl, default=true, type=bool]
     * BiasT Enable - BiasT Control
       [key=biasT_ctrl, default=true, type=bool]
     * RfNotch Enable - RF Notch Filter Control
       [key=rfnotch_ctrl, default=true, type=bool]
     * DabNotch Enable - DAB Notch Filter Control
       [key=dabnotch_ctrl, default=true, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: Tuner 1 50 ohm, Tuner 1 Hi-Z, Tuner 2 50 ohm

AlexandreRouma, 2020-08-11:
Hi
Try disabling the address sanitizer.
It seems like it sometimes makes a fuss of the way volk allocates memory.
To do that, edit CMakeLists.txt and change the line
`set(CMAKE_CXX_FLAGS "-O3 -std=c++17 -fsanitize=address -g")`
to
`set(CMAKE_CXX_FLAGS "-O3 -std=c++17")`
And then recompile.

nmaster2042, 2020-08-11:
In my case, no more issue doing this mod and trying samples rate = 125 000 or 250 000.

starvald, 2020-08-11:
that fixed it @AlexandreRouma, though for the RSP2 you'd need to add something to change which aerial is used at some point.

closed 2020-08-11 by starvald

AlexandreRouma, 2020-08-11:
@starvald I'm planning to add that but the only issue is I have no hardware to test it 😅 
I'll try to add something but I'll need someone to try it out and see if it works
I recommend you open another issue for antenna switching.