AlexandreRouma/SDRPlusPlus #11: Add antenna switching.

starvald, 2020-08-11:
SDR's such as the SDRPlay RSP2 and Duo have more than one antenna input.
ability to set the input would be handy.

nmaster2042, 2020-08-11:
SDRplays have other settings like bias tee, AM, FM DAB notch filters, ext ref clock, iq correction, agc setpoint etc.
With these settings, performances of this sdr familly can be improved
I have RSP Duo and DX to test.

nmaster2042, 2020-08-16:
With RSP Duo, for now there is two gain sliders: IFGR gain and RFGR gain.

For now, IFGR gain seems to have no effect.

horga83, 2021-02-17:
Yes the LimeSDR also requires antenna switching,  look at GQRX for an example it has antenna switching code.

horga83, 2021-02-17:
Here is some good info on soapysdr and it has antenna info in it.  I'm not a real C++ guy so I would struggle a bit but this may help a lot.

https://myriadrf.org/news/limesdr-made-simple-part-8-c-examples-soapysdr-api/

PR #76 (2021-02-23): Support selecting antenna for SoapySDR source
Partially fixes #11

[image]
closed 2021-02-23 by AlexandreRouma