AlexandreRouma/SDRPlusPlus #14: Audio drop outs with SDRPlay RSP Duo

nmaster2042, 2020-08-16:
With SDRplay RSP Duo, I get audio drop outs whatever sample rate and modulation I chose.

On the terminal, I can see line below many times:

ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred

No similar issue with RTL-SDR on the same computer

Something I see is after using RTL-SDR, less audio drop outs with RSP but still some.

AlexandreRouma, 2020-08-16:
What sample rate are you using with the RSPDuo?

nmaster2042, 2020-08-16:
Mainly 1000000 and 2000000.

But I tried all, and sample rate doesn't seem to affect the audio issue.

AlexandreRouma, 2020-08-16:
That's very weird
The input device shouldn't affect the audio.
I changed a lot of the audio system in the experimental branch, so you could maybe try that one
and see if the issue is resolved.

Here are rough compiling instructions because it's quite a bit harder then the master branch.
1. Make sure you have all dependencies (should already be the case, there aren't any new ones)
From the root directory of SDR++:
2. `mkdir build`
3. `cd build`
4. `cmake ..`
5. `make -j4`

This should build the main software, now you need to the radio module
From the root directory of SDR++ again:
2. `cd modules/radio`
2. `mkdir build`
3. `cd build`
4. `cmake ..`
5. `make -j4`

Go back to the root.
Now, copy `bandplans`, `res`, `band_colors.json`, `module_list.json` and `config.json` to the `build` directory
edit module_list.json so that the `Radio` module points to `../modules/radio/build/libradio.so` instead of what it is by default.

You should now be able to start SDR++, remember to run as root

nmaster2042, 2020-08-16:
Let's try the experiment built.

First, I managed to build it with your instructions, but I had to adapt some points.

On the SDRPlusPlus folder, I hade to modify CMakeLists.txt by replacing the line:

set(CMAKE_CXX_FLAGS "-O3 -std=c++17 -fsanitize=address -g")

by 

set(CMAKE_CXX_FLAGS "-O3 -std=c++17 -fpermissive")

because -fsanitize=address -g makes tha app crash when selecting some low samples rates and -fpermissive because without it, I get an error when building the imgui part.

On the radio module, I also edited the CMakeLists.txt to remove the -fsanitize=address -g option like on the main program.

The radio module built fine, but, on linux platform its filename is libradio.so, and it's the name that must be reported on the module_list.json.

Now all is built, and running, radio module is loaded, but audio drop outs still there.

Why should sdrpp should be started as root ?
On my system, I have no sound running as root.

I tried again with RTL-SDR and I get same audio drop outs, so, doesn't seem to be dependent of the hardware.contrary to what I was thinking. Sometimes it is ok, sometimes not, and 

I tried to play with audio settings of the radio module, if I change audio samplerate (ex from 48000 to 44100) SDRPP hngs and I have to kill it, same if a Change audio device output.

Running under ubuntu, I also tried to temporarily disable pulseaudio, but no sound, and can't change audio device, makes sdrpp to hang.

AlexandreRouma, 2020-08-16:
The address sanitizer stuff should be fixed in the next commit.
I was recommending running it as room because on my system SoapySDR doesn't find any device when not running as root (config problem on my size)

I think the issue could be with Alsa or port audio not liking the delay when the block size of the DSP doesn't match with the audio blockSize (Sadly, portaudio doesn't allow to change the block size while running).

The crashes when changing the audio sample rates are "normal" for now, I still haven't gotten around to testing the experimental code on linux.

nmaster2042, 2020-08-16:
I played with pulseaudio parameters, and found a sort of temporary workaround.

It's related to latency.

I start sdrpp like this:

$ PULSE_LATENCY_MSEC=120 ./sdrpp

I get some drop outs at starting for 2 sec and issue disappears.

I'll test more for longer time to see if it stays usable.

AlexandreRouma, 2020-08-16:
It is a fix but it's not desirable.
You'll notice the audio latency is much higher (aka, you move the VFO and it takes some time to get the new audio)

Try pulling the latest changes I just pushed, should fix the hanging when changing sample rate or output device.

Still working on fixing the cutting out issue

nmaster2042, 2020-08-16:
Yes I noticed the audio latency, it's definitively a temporary trick, I hope so.

I pulled and rebuilt your last changes, I confirm I can now change output audio device and audio sample rate.

It doesn't hang any more and it works as expected.

This doesn't affect the drop outs, I confirm too.

AlexandreRouma, 2020-12-13:
Linux audio issues mostly fixed in next release

closed 2020-12-13 by AlexandreRouma