AlexandreRouma/SDRPlusPlus #15: Can't run SDR++ on Ubuntu 20.04.1 LTS

BG7JAF, 2020-08-20:
Hi,
I have compile the program from source (7759de96da9ecb1bef242bd4e6b6a3366edfafcf), everything is ok. But when I try to  run the program, it just crash. the error message is here:

`[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,1), Device or resource busy.

[2020-08-20 15:23:57.279] [info] SDR++ v0.2.5_alpha
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
[2020-08-20 15:23:57.803] [info] Audio device open.
[2020-08-20 15:23:57.803] [info] Loading icons
[2020-08-20 15:23:57.852] [info] Loading band plans
[2020-08-20 15:23:57.852] [error] Band Plan directory does not exist
[2020-08-20 15:23:57.852] [info] Loading band plans color table
[2020-08-20 15:23:57.852] [error] Band Plan Color Table file does not exist
[2020-08-20 15:23:57.852] [info] Loading test module
[2020-08-20 15:23:57.852] [info] Ready.
[2020-08-20 15:23:57.929] [info] Changed input device: 0

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Device or resource busy.

[2020-08-20 15:23:57.941] [info] Changed sample rate: 0
AddressSanitizer:DEADLYSIGNAL
=================================================================
==3463==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x5576e08255f8 bp 0x6110001cc400 sp 0x7ffcb5e68960 T0)
==3463==The signal is caused by a READ memory access.
==3463==Hint: address points to the zero page.
    #0 0x5576e08255f7 in std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, bandplan::BandPlan_t>::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, 0ul>(std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>&, std::tuple<>&, std::_Index_tuple<0ul>, std::_Index_tuple<>) /usr/include/c++/9/bits/basic_string.h:451
    #1 0x5576e08255f7 in std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, bandplan::BandPlan_t>::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>(std::piecewise_construct_t, std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>, std::tuple<>) /usr/include/c++/9/tuple:1663
    #2 0x5576e08255f7 in void __gnu_cxx::new_allocator<std::_Rb_tree_node<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, bandplan::BandPlan_t> > >::construct<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, bandplan::BandPlan_t>, std::piecewise_construct_t const&, std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>, std::tuple<> >(std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, bandplan::BandPlan_t>*, std::piecewise_construct_t const&, std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>&&, std::tuple<>&&) /usr/include/c++/9/ext/new_allocator.h:147
    #3 0x5576e08255f7 in void std::allocator_traits<std::allocator<std::_Rb_tree_node<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, bandplan::BandPlan_t> > > >::construct<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, bandplan::BandPlan_t>, std::piecewise_construct_t const&, std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>, std::tuple<> >(std::allocator<std::_Rb_tree_node<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, bandplan::BandPlan_t> > >&, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, bandplan::BandPlan_t>*, std::piecewise_construct_t const&, std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>&&, std::tuple<>&&) /usr/include/c++/9/bits/alloc_traits.h:484
    #4 0x5576e08255f7 in void std::_Rb_tree<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, bandplan::BandPlan_t>, std::_Select1st<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, bandplan::BandPlan_t> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, bandplan::BandPlan_t> > >::_M_construct_node<std::piecewise_construct_t const&, std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>, std::tuple<> >(std::_Rb_tree_node<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, bandplan::BandPlan_t> >*, std::piecewise_construct_t const&, std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>&&, std::tuple<>&&) /usr/include/c++/9/bits/stl_tree.h:614
    #5 0x5576e08255f7 in std::_Rb_tree_node<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, bandplan::BandPlan_t> >* std::_Rb_tree<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, bandplan::BandPlan_t>, std::_Select1st<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, bandplan::BandPlan_t> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, bandplan::BandPlan_t> > >::_M_create_node<std::piecewise_construct_t const&, std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>, std::tuple<> >(std::piecewise_construct_t const&, std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>&&, std::tuple<>&&) /usr/include/c++/9/bits/stl_tree.h:631
    #6 0x5576e08255f7 in std::_Rb_tree_iterator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, bandplan::BandPlan_t> > std::_Rb_tree<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, bandplan::BandPlan_t>, std::_Select1st<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, bandplan::BandPlan_t> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, bandplan::BandPlan_t> > >::_M_emplace_hint_unique<std::piecewise_construct_t const&, std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>, std::tuple<> >(std::_Rb_tree_const_iterator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, bandplan::BandPlan_t> >, std::piecewise_construct_t const&, std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>&&, std::tuple<>&&) /usr/include/c++/9/bits/stl_tree.h:2455
[... cut ...]
AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /usr/include/c++/9/bits/basic_string.h:451 in std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, bandplan::BandPlan_t>::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, 0ul>(std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>&, std::tuple<>&, std::_Index_tuple<0ul>, std::_Index_tuple<>)
==3463==ABORTING`

How can I fix it? thx

nmaster2042, 2020-08-20:
Hi,

On the CMakefiles.txt, try to replece thie line:

set(CMAKE_CXX_FLAGS "-O3 -std=c++17 -fpermissive -fsanitize=address -g")

by

set(CMAKE_CXX_FLAGS "-O3 -std=c++17 -fpermissive")

Same for CMakefiles.txt of each module.

Then rebuild all.

AlexandreRouma, 2020-08-20:
This is a bug but disabling the adress sanitizer as mentioned above should be a temporary fix. this will be fixed in the next alpha release.

BG7JAF, 2020-08-20:
thank you, I can run it now.

closed 2020-08-20 by BG7JAF