AlexandreRouma/SDRPlusPlus #21: Building on Linux failed

BX2ABT, 2020-10-04:
System: Kubuntu 20.04
SDR++ compiles fine, when invoking ./sdrpp I get the following error

` ./sdrpp 
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
ERROR: Unable to find host: Name or service not known
[2020-10-04 21:00:00.633] [info] SDR++ v0.2.5_alpha
sdrpp: /home/hans/src/SDRPlusPlus/src/imgui/imgui_draw.cpp:1823: ImFont* ImFontAtlas::AddFontFromFileTTF(const char*, float, const ImFontConfig*, const ImWchar*): Assertion `(0) && "Could not load font file!"' failed.
Aborted (core dumped)
`
Any ideas on how to proceed?

AlexandreRouma, 2020-10-04:
Hi,
It seems like SDR++ is trying to load from a root directory that doesn't exist

This could be caused by sdrpp not being launched in the right directory (supposed to be `<repo root>/build/`)
or it's not been configured correctly.

I'll try to better document the building process. It's quite complicated at the moment because it has to load resources (like fonts and icons)

AlexandreRouma, 2020-12-13:
Issue solved in later versions.

closed 2020-12-13 by AlexandreRouma