AlexandreRouma/SDRPlusPlus #35: RX888 support

fventuri, 2020-11-13:
I just saw on twitter that you started working for support for the RX888.

I am not sure if you are aware that I wrote a low level library for Linux for both the old firmware (ExtIO dll version 0.X) and the most recent firmware version (ExtIO dll version 1.X).
The version for the old firmware is here: https://github.com/fventuri/RF103 (should work for HF; still being tested for VHF/UHF)
The version for the new firmware is here: https://github.com/fventuri/libsddc (I pushed it out only a couple of days ago; it is being tested these days).

Since they are both libraries that only depend on libusb 1.0, I thought you might be able to use them in your code (or at least take a look at the code to see how things work there).

Feel free to ask me any question about them, and congratulations for your nice SDR program!

Franco

AlexandreRouma, 2020-11-14:
Hi,

Yes I found your library, but I sadly wasn't able to get it to compile on windows :/
It seems very good though!

If you have the time, it would be great to help us develop a module using your library.
We would then have a general module for all supported hardware.

Best regards.

AlexandreRouma, 2021-03-21:
Has there been any update to libsddc? Still waiting to add support, but can't until the library has a proper repository of its own.

fventuri, 2021-03-21:
@AlexandreRouma 

A couple of updates:
- there is now a new model of that wideband SDR receiver called RX888 MKII, with improved functions and better performance; because of that the firmware that 'libsddc' used to be based on has changed (for instance the firmware is now able to autodetect the hardware being used), and therefore the version in my GitHub repo needs some changes in order to work with the new firmware
- the good news is that the libsddc code will become part and distributed with the official driver for these SDRs called 'ExtIO_sddc' (https://github.com/ik1xpv/ExtIO_sddc); Howard Su (@howard0su) has been working on integrating the code in the ExtIO_sddc driver (you can already see the 'libsddc' directory there with the source files he is working on), and hopefully we'll soon have a working library for Linux.

Franco

AlexandreRouma, 2021-03-21:
I see thank you.
As soon as a standalone library is available I'll work on adding support.
I want to avoid embedding the library in the source module for upgradability.

kpd200081, 2021-04-18:
Hello, I started to work on support of RX888 with libsddc here: https://github.com/AlexandreRouma/SDRPlusPlus/pull/97

benkyd, 2022-08-19:
Hi @fventuri, Unfortunately unless the manufacturer of the RX888 starts to take some ownership, we will not see support for the RX888 in SDR++ for some time.

You can see the full thread here https://twitter.com/ryzerth/status/1559995346968870912

closed 2022-08-19 by benkyd