AlexandreRouma/SDRPlusPlus #37: Compiling on Debian

DL5SMB, 2020-12-13:
I just tried compiling SDR++ on Debian. 

There seems to be a problem with the source code:

` 85%] Building CXX object soapy_source/CMakeFiles/soapy_source.dir/src/main.cpp.o
/home/sascha/SDRPlusPlus/soapy_source/src/main.cpp: In member function ‘void SoapyModule::selectDevice(std::__cxx11::string)’:
/home/sascha/SDRPlusPlus/soapy_source/src/main.cpp:146:29: error: ‘class SoapySDR::Device’ has no member named ‘hasIQBalanceMode’; did you mean ‘hasIQBalance’?
         hasIQBalance = dev->hasIQBalanceMode(SOAPY_SDR_RX, channelId);
                             ^~~~~~~~~~~~~~~~
                             hasIQBalance
/home/sascha/SDRPlusPlus/soapy_source/src/main.cpp: In static member function ‘static void SoapyModule::start(void*)’:
/home/sascha/SDRPlusPlus/soapy_source/src/main.cpp:246:25: error: ‘class SoapySDR::Device’ has no member named ‘setIQBalanceMode’; did you mean ‘setIQBalance’?
             _this->dev->setIQBalanceMode(SOAPY_SDR_RX, _this->channelId, _this->iqBalance);
                         ^~~~~~~~~~~~~~~~
                         setIQBalance
/home/sascha/SDRPlusPlus/soapy_source/src/main.cpp: In static member function ‘static void SoapyModule::menuHandler(void*)’:
/home/sascha/SDRPlusPlus/soapy_source/src/main.cpp:342:51: error: ‘class SoapySDR::Device’ has no member named ‘setIQBalanceMode’; did you mean ‘setIQBalance’?
                 if (_this->running) { _this->dev->setIQBalanceMode(SOAPY_SDR_RX, _this->channelId, _this->iqBalance); }
                                                   ^~~~~~~~~~~~~~~~
                                                   setIQBalance
make[2]: *** [soapy_source/CMakeFiles/soapy_source.dir/build.make:63: soapy_source/CMakeFiles/soapy_source.dir/src/main.cpp.o] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:331: soapy_source/CMakeFiles/soapy_source.dir/all] Fehler 2
make: *** [Makefile:84: all] Fehler 2
`

I did as told and then it compiled, but I am still experiencing some problems, for which I will open another case.

AlexandreRouma, 2020-12-13:
Hi,
I'll have a look, haven't had a chance to build on linux this weekend

I'll let you know once I find the issue

AlexandreRouma, 2020-12-13:
I fixed the issue, you should be able to pull the code and build as-is.
remember to edit root_dev/config.json to set the right paths for the module (changing .dll to .so and removing "Release/" from the path)

DL5SMB, 2020-12-13:
Hi Alexandre,

that did the trick, thanks for the fix. 
And also thanks for the note regarding the file extensions / paths.

The program opens now but crashes, as soon as I (try to) start the stream. But this has most probably another cause, so I'll open another case.

closed 2020-12-13 by DL5SMB