AlexandreRouma/SDRPlusPlus #38: Crash after opening plutosdr_source

DL5SMB, 2020-12-13:
Hi,

after having everything compiled, I tried to open a stream with my pluto, but as soon as hit the play button,
everything crashed.

In the log I can find various warnings and errors regarding ALSA, but there seems also to be a problem with the plutosdr_source.co

`[2020-12-13 16:50:45.509] [info] SDR++ v0.2.5_alpha
[2020-12-13 16:50:45.509] [info] Loading config
[2020-12-13 16:50:46.100] [info] Loading icons
[2020-12-13 16:50:46.111] [info] Loading band plans
[2020-12-13 16:50:46.117] [info] Loading band plans color table
[2020-12-13 16:50:46.135] [info] Loading modules
[2020-12-13 16:50:46.135] [info] Loading ./radio/radio.so
[2020-12-13 16:50:46.144] [info] Loading ./recorder/recorder.so
[2020-12-13 16:50:46.160] [info] Loading ./soapy_source/soapy_source.so
[2020-12-13 16:50:46.177] [info] Loading ./rtl_tcp_source/rtl_tcp_source.so
[2020-12-13 16:50:46.193] [info] Loading ./audio_sink/audio_sink.so
[2020-12-13 16:50:46.210] [info] Loading ./plutosdr_source/plutosdr_source.so
[2020-12-13 16:50:46.227] [info] Initializing Audio Sink (audio_sink)
[2020-12-13 16:50:46.242] [info] Initializing PlutoSDR Source (plutosdr_source)
[2020-12-13 16:50:46.259] [info] PlutoSDRSourceModule 'PlutoSDR Source': Instance created!
[2020-12-13 16:50:46.259] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2020-12-13 16:50:46.275] [info] Initializing Radio 1 (radio)
[2020-12-13 16:50:46.350] [info] Initializing Radio 2 (radio)
[2020-12-13 16:50:46.406] [info] Initializing Recorder (recorder)
[2020-12-13 16:50:46.406] [info] Initializing SoapySDR Source (soapy_source)
[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3
[2020-12-13 16:50:48.759] [info] Setting sample rate to 44100.0
[2020-12-13 16:50:48.760] [info] SoapyModule 'SoapySDR Source': Instance created!
[2020-12-13 16:50:48.760] [info] PlutoSDRSourceModule 'PlutoSDR Source': Menu Select!
[2020-12-13 16:50:48.760] [info] PlutoSDRSourceModule 'PlutoSDR Source': Tune: 0.0!
ALSA lib pcm_dsnoop.c:638:(snd_pcm_dsnoop_open) unable to open slave
ALSA lib pcm_dmix.c:1108:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_a52.c:823:(_snd_pcm_a52_open) a52 is only for playback
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_dmix.c:1108:(snd_pcm_dmix_open) unable to open slave
[2020-12-13 16:50:49.446] [info] Audio device open.
[2020-12-13 16:50:49.510] [warning] ConfigManager locked, waiting...
[2020-12-13 16:50:50.510] [warning] ConfigManager locked, waiting...
Expression 'ret' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1736
Expression 'AlsaOpen( hostApi, parameters, streamDir, &pcm )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1768
[2020-12-13 16:50:51.510] [warning] ConfigManager locked, waiting...
Expression 'ret' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1736
Expression 'AlsaOpen( hostApi, parameters, streamDir, &pcm )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1768
[2020-12-13 16:50:52.510] [warning] ConfigManager locked, waiting...
Expression 'ret' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1736
Expression 'AlsaOpen( hostApi, parameters, streamDir, &pcm )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1768
[2020-12-13 16:50:53.511] [warning] ConfigManager locked, waiting...
Expression 'ret' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1736
Expression 'AlsaOpen( hostApi, parameters, streamDir, &pcm )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1768
[2020-12-13 16:50:54.511] [warning] ConfigManager locked, waiting...
Expression 'ret' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1736
Expression 'AlsaOpen( hostApi, parameters, streamDir, &pcm )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1768
[2020-12-13 16:50:55.511] [warning] ConfigManager locked, waiting...
Expression 'ret' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1736
Expression 'AlsaOpen( hostApi, parameters, streamDir, &pcm )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1768
[2020-12-13 16:50:55.848] [info] Audio device open.
[2020-12-13 16:50:55.849] [info] PlutoSDRSourceModule 'PlutoSDR Source': Tune: 96017050.0!
[2020-12-13 16:50:55.849] [info] Ready.
[2020-12-13 16:51:02.282] [info] PlutoSDRSourceModule 'PlutoSDR Source': Tune: 97100000.0!
[2020-12-13 16:51:02.301] [info] PlutoSDRSourceModule 'PlutoSDR Source': Tune: 97100000.0!
./sdrpp: symbol lookup error: ./plutosdr_source/plutosdr_source.so: undefined symbol: iio_create_context_from_uri
sascha@think:~/SDRPlusPlus/build$ ./sdrpp 
[... 36 lines cut ...]
[2020-12-13 16:51:58.039] [warning] ConfigManager locked, waiting...
[2020-12-13 16:51:58.106] [info] Audio device open.
[2020-12-13 16:51:59.039] [warning] ConfigManager locked, waiting...
Expression 'ret' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1736
Expression 'AlsaOpen( hostApi, parameters, streamDir, &pcm )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1768
[2020-12-13 16:52:00.039] [warning] ConfigManager locked, waiting...
Expression 'ret' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1736
Expression 'AlsaOpen( hostApi, parameters, streamDir, &pcm )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1768
[2020-12-13 16:52:01.040] [warning] ConfigManager locked, waiting...
Expression 'ret' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1736
Expression 'AlsaOpen( hostApi, parameters, streamDir, &pcm )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1768
[2020-12-13 16:52:02.040] [warning] ConfigManager locked, waiting...
Expression 'ret' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1736
Expression 'AlsaOpen( hostApi, parameters, streamDir, &pcm )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1768
[2020-12-13 16:52:03.040] [warning] ConfigManager locked, waiting...
Expression 'ret' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1736
Expression 'AlsaOpen( hostApi, parameters, streamDir, &pcm )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1768
[2020-12-13 16:52:04.040] [warning] ConfigManager locked, waiting...
Expression 'ret' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1736
Expression 'AlsaOpen( hostApi, parameters, streamDir, &pcm )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1768
[2020-12-13 16:52:04.515] [info] Audio device open.
[2020-12-13 16:52:04.516] [info] PlutoSDRSourceModule 'PlutoSDR Source': Tune: 96500000.0!
[2020-12-13 16:52:04.516] [info] Ready.
**./sdrpp: symbol lookup error: ./plutosdr_source/plutosdr_source.so: undefined symbol: iio_create_context_from_uri**
`

AlexandreRouma, 2020-12-13:
Hi,
the plutosdr module isn't currently compatible with linux because of some bugs,
I recommend you use the SoapySDR source while the bug is getting fixed.

AlexandreRouma, 2021-01-01:
Bug fixed

closed 2021-01-01 by AlexandreRouma