AlexandreRouma/SDRPlusPlus #39: Debian 10.7 (buster 2020-12-05) works, but hit a few issues along the way

martin-zs, 2020-12-13:
This is not really an issue, other than the missing bit from CMakeLists.txt
But maybe some of it can be recycled by you when adding instructions for Linux:

$ uname -a
Linux SDR003 4.19.0-13-amd64 #1 SMP Debian 4.19.160-2 (2020-11-28) x86_64 GNU/Linux
$ cat /etc/debian_version 
10.7

$ mkdir src 
$ cd src
$ sudo apt install git
$ git clone https://github.com/AlexandreRouma/SDRPlusPlus
$ cd SDRPlusPlus/
$ sudo apt install build-essential
$ sudo apt install cmake
$ sudo apt install libglew-devel
$ sudo apt install libglfw3-dev
$ sudo apt install libvolk1-dev
$ sudo apt install libfftw3-dev
$ sudo apt install portaudio19-dev
$ sudo apt install libsoapysdr-dev
$ mkdir build
$ cd build
# modified CMakeLists.txt, added this line: 
target_link_libraries(sdrpp PRIVATE stdc++fs)
# to allow sdrpp to link, failed without finding "stdc++fs"
$ cmake ..
$ make -j 6
...
[ 93%] Building CXX object CMakeFiles/sdrpp.dir/src/imgui/imgui_widgets.cpp.o
[100%] Linking CXX executable sdrpp
[100%] Built target sdrpp
$ ./sdrpp 
[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3
[2020-12-13 20:36:28.327] [info] SDR++ v0.2.5_alpha
[2020-12-13 20:36:28.494] [error] Glfw Error 65543: GLX: Failed to create context: GLXBadFBConfig

$ sudo apt-get install mesa-utils
$ glxinfo | grep 'version'
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
    Max core profile version: 0.0
    Max compat profile version: 2.1
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 2.0
OpenGL version string: 2.1 Mesa 18.3.6
OpenGL shading language version string: 1.20
OpenGL ES profile version string: OpenGL ES 2.0 Mesa 18.3.6
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 1.0.16

$ grep -i glfwWindowHint ../src/*.cpp
../src/main.cpp:    glfwWindowHint(GLFW_CONTEXT_VERSION_MAJOR, 3);
../src/main.cpp:    glfwWindowHint(GLFW_CONTEXT_VERSION_MINOR, 2);
../src/main.cpp:    glfwWindowHint(GLFW_OPENGL_PROFILE, GLFW_OPENGL_CORE_PROFILE);  // 3.2+ only
../src/main.cpp:    glfwWindowHint(GLFW_OPENGL_FORWARD_COMPAT, GL_TRUE);            // Required on Mac
$

SDR++ is expecting OpenGL 3.2 at a minimum and the virtual graphics card on my virtualbox (testing it out before I install on real hardware) supports OpenGL 2.1 maximum with emulated hardware.

A slower but useful workaround is (I try new software in a VM before I install it on physical hardware):
$ LIBGL_ALWAYS_SOFTWARE=1 glxinfo | grep 'version'
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
    Max core profile version: 3.3
    Max compat profile version: 3.1
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.0
OpenGL core profile version string: 3.3 (Core Profile) Mesa 18.3.6
OpenGL core profile shading language version string: 3.30
OpenGL version string: 3.1 Mesa 18.3.6
OpenGL shading language version string: 1.40
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 18.3.6
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00
$  LIBGL_ALWAYS_SOFTWARE=1 ./sdrpp
[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3
[2020-12-13 21:06:53.488] [info] SDR++ v0.2.5_alpha
sdrpp: /home/mzs/src/SDRPlusPlus/src/imgui/imgui_draw.cpp:1823: ImFont* ImFontAtlas::AddFontFromFileTTF(const char*, float, const ImFontConfig*, const ImWchar*): Assertion `(0) && "Could not load font file!"' failed.
Aborted

$ grep -i '.ttf' ../src/*.cpp ../src/*.h
../src/frequency_select.cpp:    font = ImGui::GetIO().Fonts->AddFontFromFileTTF("res/fonts/Roboto-Medium.ttf", 42.0f);
../src/styles.h:    io.Fonts->AddFontFromFileTTF("res/fonts/Roboto-Medium.ttf", 16.0f);
cd ..
$ LIBGL_ALWAYS_SOFTWARE=1 ./build/sdrpp
[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3
[2020-12-13 21:12:14.216] [info] SDR++ v0.2.5_alpha
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib setup.c:547:(add_elem) Cannot obtain info for CTL elem (MIXER,'IEC958 Playback Default',0,0,0): No such file or directory
ALSA lib setup.c:547:(add_elem) Cannot obtain info for CTL elem (MIXER,'IEC958 Playback Default',0,0,0): No such file or directory
ALSA lib setup.c:547:(add_elem) Cannot obtain info for CTL elem (MIXER,'IEC958 Playback Default',0,0,0): No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_a52.c:823:(_snd_pcm_a52_open) a52 is only for playback
ALSA lib setup.c:547:(add_elem) Cannot obtain info for CTL elem (MIXER,'IEC958 Playback Default',0,0,0): No such file or directory
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
[2020-12-13 21:12:15.040] [info] Audio device open.
[2020-12-13 21:12:15.040] [info] Loading icons
[2020-12-13 21:12:15.100] [info] Loading band plans
[2020-12-13 21:12:15.138] [info] Loading band plans color table
[2020-12-13 21:12:15.146] [info] Loading test module
[2020-12-13 21:12:15.146] [info] Ready.
[2020-12-13 21:12:15.281] [info] Changed input device: 0

It works!

AlexandreRouma, 2020-12-13:
Hi,

Thanks for posting this, could come in useful for people having the same issue ;)

However, this seems to be the master branch which is super out of date (my bad, haven't merged from better_dsp in a while)

I recommend you try the better_dsp branch. Do note that the instructions are now invalid and I have to rewrite them (the module system changed a bit)

In the mean time, if you need help building it, I recommend joining the development discord server here:
https://discord.gg/aFgWjyD

best regards!

closed 2020-12-13 by AlexandreRouma

AlexandreRouma, 2020-12-14:
Branches have been merged and instructions have been updated.

Feel free to pull from the master branch and try building :)

Regards.