AlexandreRouma/SDRPlusPlus #43: Some features request

cropinghigh, 2020-12-14:
It will be good if these hw settings will be in SDR++:
- RTL-SDR automatic gain(available via SoapySDR)
- HackRF or other sdr's hw bandwidth(available via SoapySDR)

Also some software features:
- Variable FFT size
- Decimation(like in gqrx)
- Audio FFT and waterfall
- Stepped gain regulation
- Changing filter width by dragging it's edges on waterfall
- Zoom slider default position is max zoom, when zoom is minimal, i think better to set it to min on start
- Maybe add more accurate filter managing like in SDR#, where you can drag both sides of the filter
- Signal information like SNR when moving cursor on it

Maybe some time shifting plugin like Timeshift in SDR# included into main program.

Currently found glitches by me:
- Long starting time (> 5s)(maybe because of audio devices detection)
- When closing by default close button(cross), it prints exception in terminal like there is some errors
- Some underruns occurs

AlexandreRouma, 2020-12-14:
> Zoom slider default position is max zoom, when zoom is minimal, i think better to set it to min on start
This is a bug, i'll have a look at it in priority

cropinghigh, 2020-12-25:
Another good thing to add is changing sliders by mouse wheel, and more stepped regulators with changeable step. Like, bandwidth not 1 Hz resolution, but with selectable 1, 10, 50, 100, 500, 1000, 5000, 10000, 50000 Hz step etc. And make some sliders at least integers(like gains, don't know sdrs with non-integer gain values)