AlexandreRouma/SDRPlusPlus #45: Crash when disable/enable waterfall

wingrime, 2020-12-14:
Possible reason: heap corruption

Need investigation with valgrind

```
_int_malloc 0x00007ffff68c2da2
malloc 0x00007ffff68c3e51
operator new 0x00007ffff6abe53a
operator new[] 0x00007ffff6abe58a
ImGui::WaterFall::onResize waterfall.cpp:431
displaymenu::draw display.cpp:16
Menu::draw menu.cpp:53
drawWindow main_window.cpp:518
sdrpp_main core.cpp:269
main main.cpp:5
__libc_start_main 0x00007ffff6860152
_start 0x000055555555506e
```

AlexandreRouma, 2020-12-14:
indeed a bug of the new waterfall code,

I'll investigate, thanks for reporting

AlexandreRouma, 2020-12-14:
oops wrong button, I'll push a fix in a minute

AlexandreRouma, 2020-12-14:
Alright, I think the issue is fixed, can you pull the latest commit and config?

wingrime, 2020-12-14:
works!

closed 2020-12-14 by wingrime