AlexandreRouma/SDRPlusPlus #46: SDR++ does not start / crashes (Debian)

DL5SMB, 2020-12-24:
Hi everyone,

I just pulled a fresh copy of the new master branch and tried to compile and run it on debian.
The code was compiled, but after starting it crashes with the following message.

Any ideas what might go wrong?

I changed the paths of the libraries as I did in the old new-dsp branch, but no success.

Sascha

> sascha@think:~/SDRPlusPlus/build$ ./sdrpp
> [2020-12-24 18:12:28.448] [info] SDR++ v0.2.5_beta
> [2020-12-24 18:12:28.448] [info] Loading config
> sdrpp: /home/sascha/SDRPlusPlus/core/src/imgui/stb_image_resize.h:2394: int stbir__resize_allocated(stbir__info*, const void*, int, void*, int, int, stbir_uint32, stbir_datatype, stbir_edge, stbir_edge, stbir_colorspace, void*, size_t): Assertion `(size_t)(unsigned char*)(((unsigned char*)info->ring_buffer) + info->ring_buffer_size) == (size_t)tempmem + tempmem_size_in_bytes' failed.

AlexandreRouma, 2020-12-24:
Hi, unless you want to develop SDR++, I highly recommend you use the latest release and follow the install instructions (https://github.com/AlexandreRouma/SDRPlusPlus/releases/tag/0.2.5_beta_preview2)

If you really wanna run from source, you need to follow the instructions in the readme.
From the error message, it's clear that you didn't create a new root directory and configured everything so that it would run from the build directory.

DL5SMB, 2020-12-24:
Hi Alexandre,

thanks for the feedback and the patience. Meanwhile I checked out the precompiled  package and this works perfectly. Thanks.

However I could not leave it and had to find my mistake - so I tried it on another machine, with the same result.
But I got it working. 

My problem were 
1.) that although I created the root directory, I did not start sdrpp with the parameter "-r root_dev" and 
2.) that the path instructions were made for the previous branch I think. As we are already in the ".build/ path, the ".build" has to be removed in the suggested modules paths to make it

> "modules": [
>     "./radio/radio.so",
>     "./recorder/recorder.so",
>     "./rtl_tcp_source/rtl_tcp_source.so",
>     "./soapy_source/soapy_source.so",
>     "./audio_sink/audio_sink.so"
> ]
Its probably obvious for everyone who does this regularly, but it took a while for me to find out. Maybe it helps someone else.

Sascha

closed 2020-12-24 by DL5SMB