AlexandreRouma/SDRPlusPlus #53: No HiDPI scaling

wingrime, 2020-12-28:
on 4k display UI is simply small

AlexandreRouma, 2020-12-28:
Hello,

Yes, there currently isn't a scaling option.
It's pretty hard to add because of the waterfall

I'll try to add it after the exams when I have a bit more time

bubnikv, 2021-06-13:
Good luck with your exams.

BTW in our PrusaSlicer
https://github.com/prusa3d/PrusaSlicer/
we are using ImGUI on Windows / Linux / OSX and we have implemented HiDPI support successfully. You may get some inspiration from there, I may be of some help.

abougouffa, 2021-08-04:
Hello @AlexandreRouma , 

I don't have any experience with `ImGui`, but I think this can help [https://github.com/ocornut/imgui/blob/master/docs/FAQ.md#q-how-should-i-handle-dpi-in-my-application](https://github.com/ocornut/imgui/blob/master/docs/FAQ.md#q-how-should-i-handle-dpi-in-my-application)

shuyuan-liu, 2021-08-31:
SatDump has recently implemented manual DPI scaling to change font sizes, code [here](https://github.com/altillimity/SatDump/blob/9a144d894661d5b8eeffcd346e2aaef5dd5ecf07/src-interface/style.cpp). Only text and associated widgets are scaled, not any graphics or plots. I only found code setting the font sizes though so I'm not sure how the widgets are scaled with them.

> @AlexandreRouma: It's pretty hard to add because of the waterfall

I think it would actually be fine to not scale the contents of the waterfall, because the user can always zoom in by themselves anyway.

AlexandreRouma, 2021-08-31:
> I think it would actually be fine to not scale the contents of the waterfall, because the user can always zoom in by themselves anyway.

the issue is the text on the wtaerfall that has to be scaled and changes a lot of the areas, not the waterfall data itself.

Satdump has a much simpler GUI compared to SDR++ with a dozen of custom drawn widgets

AlexandreRouma, 2022-07-16:
implemented a while back

closed 2022-07-16 by AlexandreRouma