AlexandreRouma/SDRPlusPlus #54: Could not load any sources except rtl tcp on windows

wingrime, 2020-12-28:
using recent binary release

AlexandreRouma, 2020-12-28:
Hi,

do you have PothosSDR installed on your system?
Also, could you provide the start logs?

Regards,
Alex

wingrime, 2020-12-28:
‘’’

[2020-12-28 18:13:33.197] [info] Loading ./modules/airspyhf_source.dll
[2020-12-28 18:13:33.199] [error] Couldn't load ./modules/airspyhf_source.dll.
[2020-12-28 18:13:33.199] [info] Loading ./modules/audio_sink.dll
[2020-12-28 18:13:33.211] [info] Loading ./modules/plutosdr_source.dll
[2020-12-28 18:13:33.228] [error] Couldn't load ./modules/plutosdr_source.dll.
[2020-12-28 18:13:33.229] [info] Loading ./modules/radio.dll
[2020-12-28 18:13:33.284] [info] Loading ./modules/recorder.dll
[2020-12-28 18:13:33.286] [info] Loading ./modules/rtl_tcp_source.dll
[2020-12-28 18:13:33.334] [info] Loading ./modules/soapy_source.dll
[2020-12-28 18:13:33.336] [error] Couldn't load ./modules/soapy_source.dll.
[2020-12-28 18:13:33.336] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2020-12-28 18:13:33.386] [error] Module 'airspyhf_source' doesn't exist
[2020-12-28 18:13:33.386] [info] Initializing Audio Sink (audio_sink)
[2020-12-28 18:13:33.387] [info] Initializing PlutoSDR Source (plutosdr_source)
[2020-12-28 18:13:33.393] [error] Module 'plutosdr_source' doesn't exist
[2020-12-28 18:13:33.393] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2020-12-28 18:13:33.410] [info] Initializing Radio (radio)
[2020-12-28 18:13:33.464] [info] Initializing Recorder (recorder)
[2020-12-28 18:13:33.465] [info] Initializing SoapySDR Source (soapy_source)
[2020-12-28 18:13:33.465] [error] Module 'soapy_source' doesn't exist
[2020-12-28 18:13:33.466] [info] RTLTCPSourceModule 'RTL-TCP Source': Menu Select!
[2020-12-28 18:13:33.467] [info] RTLTCPSourceModule 'RTL-TCP Source': Tune: 0.0!
[2020-12-28 18:13:33.476] [info] RTLTCPSourceModule 'RTL-TCP Source': Tune: 100000000.0!
[2020-12-28 18:13:33.477] [info] Ready.

‘’’

AlexandreRouma, 2020-12-28:
I see, It hink I might have forgotten adependency that's needed by some of the library used.

The quickest fix is to install pothosdr

wingrime, 2020-12-28:
it was missing PothosSDR after install situation is better 
soapy works,  but no direct hackrf and rtl-sdr modules

wingrime, 2020-12-28:
another issue, by default audio is None

wingrime, 2020-12-28:
it is possible to bundle soapySDR in binary release?

AlexandreRouma, 2020-12-28:
> soapy works, but no direct hackrf and rtl-sdr modules
Yes there curently isn't a RTL-SDR module and the hackrf one is quite broken :/

> another issue, by default audio is None
yes, this is normal (because of the modular nature of sinks it's not sure which sinks it'll have at startup)
I might change it later if it's a people prefer to have the first non-Null sink as default instead

> it is possible to bundle soapySDR in binary release?
Sadly, it's not. SoapySDR needs to be installed in a specific way to work.

Regards.

closed 2020-12-28 by wingrime