AlexandreRouma/SDRPlusPlus #56: worser

wingrime, 2020-12-28:


wingrime, 2020-12-28:
[image]
[image]

wingrime, 2020-12-28:
[audio.zip](https://github.com/AlexandreRouma/SDRPlusPlus/files/5747527/audio.zip)

AlexandreRouma, 2020-12-28:
Hi,

SDRUno has audio filtering and EQ while SDR++ currently does not.

wingrime, 2020-12-28:
Noise Reduction is tuned off, 
I can't find where EQ  can be trun on or off. 

I can also compare with CubicSDR and it also better (last time I tried), and I still think something wrong with demodulation itself

AlexandreRouma, 2020-12-28:
Ah, I think I might have identified the issue.
I think my resampling filters have an issue. Lots of harmonics that shouldn't be there

I'll mark this issue as a bug.

AlexandreRouma, 2021-01-01:
Bug should now be fixed

AlexandreRouma, 2021-02-10:
Closing for inactivity, please reopen if the issue still exists

closed 2021-02-10 by AlexandreRouma