AlexandreRouma/SDRPlusPlus #58: DLL fail to load on Windows x64

JoNoci, 2021-01-01:
I have installed sdrpp_v0.2.5_beta_preview_x64.zip. When executing I get errors as indicated in the attached screen dump.
I have also downloaded the source code, rebuilt it and get the same errors when executing.
Please could you advise how to get past this problem
[image]
[image]

AlexandreRouma, 2021-01-01:
Hi,
you're missing PothosSDR or you don't have it in your path

Regards.

JoNoci, 2021-01-01:
Many thanks... my error. I have the Airspy Mini dongle. The software is
trying to start Airspy HF+ and failing. Does the software support Airspy
Mini? How do I select it?
Regards
Joe

AlexandreRouma, 2021-01-01:
the Airspy HF+ source if for the HF+ Discovery and HF+ Dual.
I'm currently writing a source for the Airspy R2 and Mini, but in that mean time I recommend using the SoapySDR source.
Your device should appear in the list of the SoapySDR source menu. If you, click refresh or restart SDR++

JoNoci, 2021-01-01:
Got it... thanks for the support
Joe

closed 2021-01-01 by AlexandreRouma