AlexandreRouma/SDRPlusPlus #61: Software crashes if startup with showWaterfall set false

JoNoci, 2021-01-10:
SDR++ v0.2.5_beta (Dec 24 2020). If I execute this software, disable view of the waterfall, and exit - thereafter the software 'crashes' when I try execute it again. The only way to get it to execute again is to manually edit the config.json to set showWaterfall to true. Hence it appears that the software has a problem if it is started with showWaterfall set false.

AlexandreRouma, 2021-01-10:
Hi

I'm not able to reproduce this on windows, what OS are you using?

nmaster2042, 2021-01-10:
Hi, tried this on ubuntu, sdrpp built from sources, no issue.

dd5fz, 2021-01-10:
This may well be the reason why my test on a Raspberry Pi 4 is now crashing (issue #62).

JoNoci, 2021-01-10:
I am using Windows 10. I get the problem with both the build I downloaded
as a pre-built build and also if I recompile the code myself. Note that the
'crash' only happens when you start the SDR using the START button top left
- ie the code starts executing, I see just the window for the FFT display
is allocated - then START --> then it crashes

AlexandreRouma, 2021-01-10:
> Note that the
'crash' only happens when you start the SDR using the START button top left. - ie the code starts executing, I see just the window for the FFT display
is allocated - then START --> then it crashes

I See, this is an important detail
I'll try to repro this in a minute

AlexandreRouma, 2021-01-10:
Can repro the bug, I think I know where the issue is

AlexandreRouma, 2021-01-11:
The bug should be fixed.
Could you pull and build the latest commit to confirm?

JoNoci, 2021-01-11:
Many thanks - it is sorted.
Joe

closed 2021-01-11 by AlexandreRouma