AlexandreRouma/SDRPlusPlus #62: SDR++ v0.2.5_beta PREVIEW #2 on Raspberry Pi

dd5fz, 2021-01-10:
I just compiled SDR++ v0.2.5_beta PREVIEW #2 on my Raspberry Pi 4 and while the compile appears to have worked, I noticed a few things:

1) The workaround suggested in issue #39 was also necessary on the Raspberry Pi 4. The SDR hardware is a NeSDR Smart (RTL-SDR) and the Linux version on my Raspberry Pi is:

```
toby@raspberrypi:~/sdr++/SDRPlusPlus-0.2.5_beta_preview2 $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

toby@raspberrypi:~/sdr++/SDRPlusPlus-0.2.5_beta_preview2 $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster

toby@raspberrypi:~/sdr++/SDRPlusPlus-0.2.5_beta_preview2 $ hostnamectl
   Static hostname: raspberrypi
         Icon name: computer
        Machine ID: 6eba8ddf347342c1995723fba1eb67c0
           Boot ID: 7ad6103e527f4e22aa4730ca50240969
  Operating System: Raspbian GNU/Linux 10 (buster)
            Kernel: Linux 5.4.83-v7l+
      Architecture: arm
toby@raspberrypi:~/sdr++/SDRPlusPlus-0.2.5_beta_preview2 $ 
```

2) The CPU load on the Raspberry Pi 4, overclocked to 1.8Ghz, is fairly high (above 85%) and I am getting a series of 
`ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred` 
errors. 

Also the software appears to have problems with the high sample rates, e.g. at 2.048 msps the audio becomes very stocky and distorted. In comparison CubicSDR runs cleanly at 2.56 msps and causes a CPU load of about 45 to 50%.

3) I just tried to restart sdr++ and am now getting the following messages:

```
toby@raspberrypi:~/sdr++/SDRPlusPlus-0.2.5_beta_preview2 $ LIBGL_ALWAYS_SOFTWARE=1 ./build/sdrpp -r root_dev
[2021-01-10 18:39:31.962] [info] SDR++ v0.2.5_beta
[2021-01-10 18:39:31.962] [info] Loading config
[2021-01-10 18:39:32.709] [info] Loading icons
[2021-01-10 18:39:32.961] [info] Loading band plans
[2021-01-10 18:39:32.994] [info] Loading band plans color table
[2021-01-10 18:39:33.043] [info] Loading modules
[2021-01-10 18:39:33.043] [info] Loading ./build/radio/radio.so
[2021-01-10 18:39:33.081] [info] Loading ./build/recorder/recorder.so
[2021-01-10 18:39:33.113] [info] Loading ./build/rtl_tcp_source/rtl_tcp_source.so
[2021-01-10 18:39:33.143] [info] Loading ./build/soapy_source/soapy_source.so
[2021-01-10 18:39:33.174] [info] Loading ./build/audio_sink/audio_sink.so
[2021-01-10 18:39:33.202] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2021-01-10 18:39:33.229] [error] Module 'airspyhf_source' doesn't exist
[2021-01-10 18:39:33.229] [info] Initializing Audio Sink (audio_sink)
[2021-01-10 18:39:33.271] [info] Initializing PlutoSDR Source (plutosdr_source)
[2021-01-10 18:39:33.298] [error] Module 'plutttosdr_source' doesn't exist
[2021-01-10 18:39:33.298] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2021-01-10 18:39:33.325] [info] Initializing Radio (radio)
[2021-01-10 18:39:33.595] [info] Initializing Recorder (recorder)
[2021-01-10 18:39:33.621] [info] Initializing SoapySDR Source (soapy_source)
[ERROR] SoapySSDPEndpoint::sendTo(udp://[ff02::c]:1900) = -1
  sendto(udp://[ff02::c]:1900) [99: Cannot assign requested address]
Found Rafael Micro R820T tuner
[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3
Found Rafael Micro R820T tuner
[2021-01-10 18:39:36.474] [info] Setting sample rate to 1024000.0
[2021-01-10 18:39:36.476] [info] SoapyModule 'SoapySDR Source': Instance created!
[2021-01-10 18:39:36.476] [info] SoapyModule 'SoapySDR Source': Menu Select!
[2021-01-10 18:39:36.478] [info] SoapyModule 'SoapySDR Source': Tune: 0.0!
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_a52.c:823:(_snd_pcm_a52_open) a52 is only for playback
ALSA lib conf.c:5014:(snd_config_expand) Unknown parameters {AES0 0x6 AES1 0x82 AES2 0x0 AES3 0x2  CARD 0}
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM iec958:{AES0 0x6 AES1 0x82 AES2 0x0 AES3 0x2  CARD 0}
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
[AVAudioResampleContext @ 0x2f8450] Value 160.000000 for parameter 'phase_shift' out of range [0 - 30]
[AVAudioResampleContext @ 0x2fa060] Value 320.000000 for parameter 'phase_shift' out of range [0 - 30]
[AVAudioResampleContext @ 0x2fa060] Value 640.000000 for parameter 'phase_shift' out of range [0 - 30]
[2021-01-10 18:39:36.964] [warning] ConfigManager locked, waiting...
[2021-01-10 18:39:37.083] [info] Audio device open.
[2021-01-10 18:39:37.132] [info] SoapyModule 'SoapySDR Source': Tune: 101709600.0!
[2021-01-10 18:39:37.133] [info] Ready.
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[INFO] Using format CF32.
Allocating 15 zero-copy buffers
[2021-01-10 18:39:45.996] [info] SoapyModule 'SoapySDR Source': Start!
[2021-01-10 18:39:46.047] [info] SoapyModule 'SoapySDR Source': Tune: 101709600.0!
ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred
Floating point exception
toby@raspberrypi:~/sdr++/SDRPlusPlus-0.2.5_beta_preview2 $
```

AlexandreRouma, 2021-01-10:
Hi,
try building from the master branch
The DSP performance issues have been fixed but are not present in release #2.

dd5fz, 2021-01-10:
I'll try doing that in the next couple days and let you know how it goes.

sdramateur, 2021-02-21:
When start BETA #3 below error

[2021-02-21 01:29:37.755] [info] SDR++ v0.2.5_beta
[2021-02-21 01:29:37.759] [info] Loading config
[2021-02-21 01:29:38.303] [info] Loading icons
[2021-02-21 01:29:38.334] [info] Loading band plans
[2021-02-21 01:29:38.349] [info] Loading band plans color table
[2021-02-21 01:29:38.354] [info] Loading modules
[2021-02-21 01:29:38.354] [info] Loading ./modules/airspyhf_source.dll
[2021-02-21 01:29:38.399] [info] Loading ./modules/airspy_source.dll
[2021-02-21 01:29:38.418] [info] Loading ./modules/audio_sink.dll
[2021-02-21 01:29:38.439] [info] Loading ./modules/plutosdr_source.dll
[2021-02-21 01:29:38.471] [info] Loading ./modules/radio.dll
[2021-02-21 01:29:38.482] [info] Loading ./modules/recorder.dll
[2021-02-21 01:29:38.494] [info] Loading ./modules/rtl_sdr_source.dll
[2021-02-21 01:29:40.985] [error] Couldn't load ./modules/rtl_sdr_source.dll.
[2021-02-21 01:29:40.986] [info] Loading ./modules/rtl_tcp_source.dll
[2021-02-21 01:29:41.001] [info] Loading ./modules/soapy_source.dll
[2021-02-21 01:29:41.033] [info] Initializing Airspy Source (airspy_source)
[2021-02-21 01:29:41.062] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2021-02-21 01:29:41.076] [info] Initializing Audio Sink (audio_sink)
[2021-02-21 01:29:41.323] [info] Initializing HackRF Source (hackrf_source)
[2021-02-21 01:29:41.325] [error] Module 'hackrf_source' doesn't exist
[2021-02-21 01:29:41.325] [info] Initializing PlutoSDR Source (plutosdr_source)
[2021-02-21 01:29:41.327] [info] PlutoSDRSourceModule 'PlutoSDR Source': Instance created!
[2021-02-21 01:29:41.327] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2021-02-21 01:29:41.328] [info] Initializing Radio (radio)
[2021-02-21 01:29:41.353] [info] Initializing Recorder (recorder)
[2021-02-21 01:29:41.355] [info] Initializing SoapySDR Source (soapy_source)
←[1m←[31m[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/uhdSupport.dll)
  LoadLibrary() failed: Kan opgegeven procedure niet vinden.
←[0m
[2021-02-21 01:29:43.324] [info] Setting sample rate to 6000000.0
[2021-02-21 01:29:43.328] [info] SoapyModule 'SoapySDR Source': Instance created!
[2021-02-21 01:29:43.333] [info] Loading color maps
[2021-02-21 01:29:43.336] [info] Loading ./res/colormaps/classic.json
[2021-02-21 01:29:43.339] [info] Loading ./res/colormaps/greyscale.json
[2021-02-21 01:29:43.340] [info] Loading ./res/colormaps/inferno.json
[2021-02-21 01:29:43.343] [info] Loading ./res/colormaps/magma.json
[2021-02-21 01:29:43.347] [info] Loading ./res/colormaps/plasma.json
[2021-02-21 01:29:43.351] [info] Loading ./res/colormaps/turbo.json
[2021-02-21 01:29:43.354] [info] Loading ./res/colormaps/viridis.json
[2021-02-21 01:29:43.372] [info] SoapyModule 'SoapySDR Source': Menu Select!
[2021-02-21 01:29:43.376] [info] SoapyModule 'SoapySDR Source': Tune: 0.0!
[2021-02-21 01:29:43.546] [info] Audio device open.
[2021-02-21 01:29:43.566] [info] SoapyModule 'SoapySDR Source': Tune: 427787987.0!
[2021-02-21 01:29:43.567] [info] Ready.
[2021-02-21 01:29:43.568] [info] SoapyModule 'SoapySDR Source': Tune: 427787987.0!

AlexandreRouma, 2021-04-30:
Closing for inactivity

closed 2021-04-30 by AlexandreRouma