AlexandreRouma/SDRPlusPlus #63: Playback of recorded baseband

JoNoci, 2021-01-11:
The software allows the recording of the IQ data to a .wav file. Is there a way that this data can be 'played back' into the software for post analysis - ie to run the existing functionality as is but sourcing the IQdata from the .wav file?

AlexandreRouma, 2021-01-11:
not yet.
The module is in development but is currently unusable.

JoNoci, 2021-01-12:
As this is something I am really keen on,would it be possible to email me
the module in its current status. If I manage to get it working I will then
send it back to you?
Regards
Joe

AlexandreRouma, 2021-02-10:
Hi, a crude baseband file player has been added. it's currently a bit unstable but works fine most of the time

AlexandreRouma, 2021-04-14:
the file player is now fixed and enabled by default. Make sure to delete the existing config file or update the config manually to enable the module.

closed 2021-04-14 by AlexandreRouma