AlexandreRouma/SDRPlusPlus #65: SoapySDR w/ BladeRF no Audio on some bandwidth selections

alphafox02, 2021-02-06:
Just taking SDRPlusPlus for a test spin. Built from source on Ubuntu 20.04.

GUI comes up great, I can see the BladeRF and it does work with the default bandwidth selections. What's interesting is that if I select the 4-12M range I get no audio, at least on WFM in the FM range. Before and after the 4-12 range, all the way up to 60+M I can get audio.

I then close sdr++ and open up CubicSDR built from source and select the same 4-12 range and I get audio. No idea the difference, but figured I'd share. I've not tested SDR++ with any other SDRs yet.

AlexandreRouma, 2021-02-06:
Hello,
could you show the logs and screenshot the software when you have the issue with the audio?

alphafox02, 2021-03-21:
Sorry I never got back to you, but your Falcon decoder caught my attention again so I came back to rebuild. While I couldn’t get the latest git to completely build, the Falcon part was failing, I was able to build “around it” and get most everything working in Linux.  

I’ll come back and check tomorrow to see if there’s some more code and or a release uploaded and check again. Judging by your posts I found where you got the rsp1a and bladerf in hand I’m sure It’ll all be fixed up soon. I have the same equipment here to test once the new release is out.

AlexandreRouma, 2021-05-13:
The bladerf now has its own module (although not finished) and the issue is solved.

closed 2021-05-13 by AlexandreRouma

alphafox02, 2021-05-13:
Perfect, I’ll rebuild and include it.