AlexandreRouma/SDRPlusPlus #70: RTLSDR/AIRSPY win10

sdramateur, 2021-02-15:
Hi,  SDR++  not find RTL-SDR nor AIRSPY window error not find module  only the Pluto
Installed pothos  cubicSDR is running

[2021-02-15 15:11:06.469] [info] SDR++ v0.2.5_beta
[2021-02-15 15:11:06.471] [info] Loading config
[2021-02-15 15:11:06.947] [info] Loading icons
[2021-02-15 15:11:06.975] [info] Loading band plans
[2021-02-15 15:11:06.977] [info] Loading band plans color table
[2021-02-15 15:11:06.982] [info] Loading modules
[2021-02-15 15:11:06.982] [info] Loading ./modules/airspyhf_source.dll
[2021-02-15 15:11:06.986] [error] Couldn't load ./modules/airspyhf_source.dll.
[2021-02-15 15:11:06.986] [info] Loading ./modules/airspy_source.dll
[2021-02-15 15:11:06.989] [error] Couldn't load ./modules/airspy_source.dll.
[2021-02-15 15:11:06.990] [info] Loading ./modules/audio_sink.dll
[2021-02-15 15:11:06.992] [info] Loading ./modules/plutosdr_source.dll
[2021-02-15 15:11:06.997] [info] Loading ./modules/radio.dll
[2021-02-15 15:11:06.999] [info] Loading ./modules/recorder.dll
[2021-02-15 15:11:07.001] [info] Loading ./modules/rtl_tcp_source.dll
[2021-02-15 15:11:07.003] [info] Loading ./modules/soapy_source.dll
[2021-02-15 15:11:07.005] [error] Couldn't load ./modules/soapy_source.dll.
[2021-02-15 15:11:07.006] [info] Initializing Airspy Source (airspy_source)
[2021-02-15 15:11:07.007] [error] Module 'airspy_source' doesn't exist
[2021-02-15 15:11:07.008] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2021-02-15 15:11:07.009] [error] Module 'airspyhf_source' doesn't exist
[2021-02-15 15:11:07.009] [info] Initializing Audio Sink (audio_sink)
[2021-02-15 15:11:07.258] [info] Initializing HackRF Source (hackrf_source)
[2021-02-15 15:11:07.260] [error] Module 'hackrf_source' doesn't exist
[2021-02-15 15:11:07.260] [info] Initializing PlutoSDR Source (plutosdr_source)
[2021-02-15 15:11:07.262] [info] PlutoSDRSourceModule 'PlutoSDR Source': Instance created!
[2021-02-15 15:11:07.262] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2021-02-15 15:11:07.263] [info] Initializing Radio (radio)
[2021-02-15 15:11:07.287] [info] Initializing Recorder (recorder)
[2021-02-15 15:11:07.289] [info] Initializing SoapySDR Source (soapy_source)
[2021-02-15 15:11:07.291] [error] Module 'soapy_source' doesn't exist
[2021-02-15 15:11:07.292] [info] Loading color maps
[2021-02-15 15:11:07.294] [info] Loading ./res/colormaps/classic.json
[2021-02-15 15:11:07.295] [info] Loading ./res/colormaps/greyscale.json
[2021-02-15 15:11:07.296] [info] Loading ./res/colormaps/inferno.json
[2021-02-15 15:11:07.298] [info] Loading ./res/colormaps/magma.json
[2021-02-15 15:11:07.299] [info] Loading ./res/colormaps/plasma.json
[2021-02-15 15:11:07.300] [info] Loading ./res/colormaps/turbo.json
[2021-02-15 15:11:07.301] [info] Loading ./res/colormaps/viridis.json
[2021-02-15 15:11:07.322] [info] PlutoSDRSourceModule 'PlutoSDR Source': Menu Select!
[2021-02-15 15:11:07.323] [info] PlutoSDRSourceModule 'PlutoSDR Source': Tune: 0.0!
[2021-02-15 15:11:07.346] [info] PlutoSDRSourceModule 'PlutoSDR Source': Tune: 100000000.0!
[2021-02-15 15:11:07.347] [info] Ready.

AlexandreRouma, 2021-02-15:
hello,
the pre-release #3 doesn't have the RTL-SDR module yet. 
I will release a version with in tomorrow, but in the mean time, here is a build that has it:
[sdrpp_windows_x64.zip](https://github.com/AlexandreRouma/SDRPlusPlus/files/5984101/sdrpp_windows_x64.zip)

For the airspy issue, I'm not too sure, are you able to use it with the Soapy module? 

I think the most probable issue is that libairspy and libairspyhf are out of date. So I would recommend trying to update pothos and using the build included here.

sdramateur, 2021-02-16:
First:   SDR#  is working for many years   CubicSDR + GQRX are working 
SDR++
Airspy is detected  but seem not receive any signal
RTLSDR not doing

Startup SDR++ i get a popup ERROR libusb_set_option not found in Program File\PothosSDR\bin\uhd.dll

AlexandreRouma, 2021-02-16:
Likely a soapy bug, sdr++ doesn't use UHD at all

sdramateur, 2021-02-16:
RTLSDR when tuning, there is some shorts "hickups" like the screen freezes its not smooth tuning

AlexandreRouma, 2021-02-16:
> RTLSDR when tuning, there is some shorts "hickups" like the screen freezes its not smooth tuning

This is normal for a RTL-SDR. The RTL cannot tune while streaming samples. This means the stream has to be cut every time you tune resulting in inconsistancies

sdramateur, 2021-02-17:
Your example picture  show multi radio support?  (radio1 + radio2)  How do i setup RTLSDR + AIRSPY?
It seem settings also record path not saved after close?

AlexandreRouma, 2021-02-17:
Multi VFS is not multi device. It can only receive from one device at once, but can decodes several parts of the spectrum.

sdramateur, 2021-02-18:
But How do i setup MULTI VFO?

AlexandreRouma, 2021-02-19:
you need to edit the config.json file and add an entry in the "moduleInstance" parameter. After that, you need to add an entry in the "menuOrder" parameter to decide where to show the menu.

sdramateur, 2021-02-21:
When start BETA #3 below error

[2021-02-21 01:29:38.494] [info] Loading ./modules/rtl_sdr_source.dll
[2021-02-21 01:29:40.985] [error] Couldn't load ./modules/rtl_sdr_source.dll.
[2021-02-21 01:29:40.986] [info] Loading ./modules/rtl_tcp_source.dll

[2021-02-21 01:29:41.323] [info] Initializing HackRF Source (hackrf_source)
[2021-02-21 01:29:41.325] [error] Module 'hackrf_source' doesn't exist
[2021-02-21 01:29:41.325] [info] Initializing PlutoSDR Source (plutosdr_source)

[2021-02-21 01:29:41.355] [info] Initializing SoapySDR Source (soapy_source)
←[1m←[31m[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/uhdSupport.dll)
  LoadLibrary() failed: Kan opgegeven procedure niet vinden.
←[0m
[2021-02-21 01:29:43.324] [info] Setting sample rate to 6000000.0

sdramateur, 2021-02-23:
It here a typo thingy  ?
./modules/rtl_sdr_source.dll.  <== the dot?

AlexandreRouma, 2021-02-23:
You either misstyped the filename in the config or you don't have librtlsdr

sdramateur, 2021-02-23:
Nope,  just install the BETA #3  and notice the startup screen give 2 ERRORS  and i see the  . DOT in the end  while other startup liners not have the . DOT

sdramateur, 2021-05-08:
Version 0.2.5. BETA Feb 2021 is working but your latest windows,  stuck with "loading audiosink.dll"

AlexandreRouma, 2021-05-08:
it's old, use the latest nightly build from here: https://github.com/AlexandreRouma/SDRPlusPlus/actions/runs/812047935
(scroll down and select sdrpp_windows_x64)

sdramateur, 2021-05-08:
your latest windows nightly build download yesterday  after install its stuck with loading "audiosink.dll" and "recorder.dll" 
When copy these audiosink.dll" and "recorder.dll from the older release 0.2.5Beta februari,  than nightly build download from yesterday, is startup without error

closed 2022-07-12 by AlexandreRouma