AlexandreRouma/SDRPlusPlus #71: Shared library libvolk.so.1.4 not present on Ubuntu 20.04 and 20.10

peterg10, 2021-02-16:
Installed the prerequisite package list (including`libvolk2-dev`) but when I run it, it's missing the v1 libvolk library.....

`sdrpp: error while loading shared libraries: libvolk.so.1.4: cannot open shared object file: No such file or directory`

I have a Debian 10 system and I see that it _does_ include the v1 library (`libvolk1-dev/stable`) so this is specific to Ubuntu 20.04 and 20.10.

AlexandreRouma, 2021-02-16:
Hello,
This is indeed a ubuntu issue. It seems that some versions have v1 and some have v2. As a default I decided to go with the older version as it should technically be available to more people.

In this case, I recommend you build the software from source. It's pretty easy since you already have the dependenceis. Here is how you should do it:

```
git clone https://github.com/AlexandreRouma/SDRPlusPlus
cd SDRPlusPlus
mkdir build
cd build
cmake ..
make -j$(nproc)
sudo make install
```

peterg10, 2021-02-16:
Thanks that worked. When building from source `librtlsdr-dev` is also needed.

AlexandreRouma, 2021-02-16:
yup, forgot to add that to the readme, thanks for mentioning it.

I'll close this issue as it seems to have been resolved.

closed 2021-02-16 by AlexandreRouma

stdevPavelmc, 2021-02-20:
This bug still exist and has an easy fix.

Fresh Ubuntu 20.04 LTS updated to the day, sdrpp v0.2.5_beta PREVIEW#3

```sh
pavel@agatha:~/$ sdrpp 
sdrpp: error while loading shared libraries: libvolk.so.1.4: cannot open shared object file: No such file or directory
pavel@agatha:~/$ volk
volk-config-info  volk_modtool      volk_profile
pavel@agatha:~/$
```

FIX: Linking the actual volk library works, see transcript below

```sh
pavel@agatha:~/$ sdrpp
sdrpp: error while loading shared libraries: libvolk.so.1.4: cannot open shared object file: No such file or directory
pavel@agatha:~/$ sudo ln -s /usr/lib/x86_64-linux-gnu/libvolk.so /usr/lib/x86_64-linux-gnu/libvolk.so.1.4
[sudo] contraseña para pavel: 
pavel@agatha:~/$ sdrpp
[2021-02-20 09:20:03.686] [info] SDR++ v0.2.5_beta
[2021-02-20 09:20:03.686] [warning] Root directory /home/pavel/.config/sdrpp does not exist, creating it
[2021-02-20 09:20:03.686] [info] Loading config
[2021-02-20 09:20:03.686] [warning] Config file '/home/pavel/.config/sdrpp/config.json' does not exist, creating it
[2021-02-20 09:20:03.923] [info] Loading icons
[2021-02-20 09:20:03.936] [info] Loading band plans
[2021-02-20 09:20:03.941] [info] Loading band plans color table
[2021-02-20 09:20:03.950] [info] Loading modules
[2021-02-20 09:20:03.950] [info] Loading /usr/lib/sdrpp/plugins/soapy_source.so
[...]
```

AlexandreRouma, 2021-02-23:
This is not a good fix.
Linking a recent version to an older version to make the software think it has the version it wants will cause problems in case libvolk deprecates certain functions.

It's a janky solution that may work at the moment but I recommend building from source. THat way, in the envent that some functions are unavailable, it'll clearly show an error during compilation and make it easier to solve.