AlexandreRouma/SDRPlusPlus #72: decoders?

sdramateur, 2021-02-16:
(not have you email so do this)
Will be possible in near future decoders like gr-gsm, LTE, Iridium, tetra (linux version) and many others using SDR++ as frontend?  This will be a very, very large step forward and leave others SDRsoftware very far behind.

AlexandreRouma, 2021-02-16:
yes, they're planned. I'm working with several people to write those. LRPT and APT are the first on the line. For iridium and inmarsat voice, I'm afraid it likely isn't legal so I personally won't be writing those, but if someone else wants to they're free to do so.

sdramateur, 2021-02-16:
Those are existing linux  or windows only need to be ported as plugin
Something else,  the LINUX/Debian pack seem rather big size, Why?

AlexandreRouma, 2021-02-16:
> Those are existing linux or windows only need to be ported as plugin

It's not as easy as that. Just because they exist doesn't mean it's simple to intergrate as a module. Especially to make sure it's cross platform. They largely have to be completely reimplemented

> Something else, the LINUX/Debian pack seem rather big size when installed, bigger as GQRX or CUBICsdr
I don't think it's really that big, but the main thing is, it's compiled with its own DSP libary. If you factored in GNU Radio for GQRX or liquid dsp for Cubic, you would notice it's just as big if not bigger. Also, SDR++ is compiled as -O3 which optimises performance at the cost of size