AlexandreRouma/SDRPlusPlus #73: HackRF Antenna Power (bias-t)

bluegizmo83, 2021-02-17:
PLEASE add Antenna Power for the HackRF if it's not already included (didn't see it in your GitHub code)! This is the one thing I really hated about SDRSharp. The HackRF has an antenna power feature, and other programs use it, but SDRSharp never did.

AlexandreRouma, 2021-02-17:
Hello,
The HackRF module is currently non-functional.
Bias-T functionality will be added when I get back to it.

Regards,
Alex

stdevPavelmc, 2021-02-20:
> The HackRF module is currently non-functional.

I was to file a bug for this, Ubuntu 20.04 LTS, can see the HackRF, glad I searched first.

AlexandreRouma, 2021-04-12:
Bias-T and the hackrf module should now work. Just make sure to disable SoapySDR or SoapyHackRF because of a bug in libhackrf.

closed 2021-04-12 by AlexandreRouma