AlexandreRouma/SDRPlusPlus #74: Core dump on startup, Ubuntu 20.10 (x86/64 / amd64)

pseudofunizer, 2021-02-17:
Hi!

I just downloaded and compiled the Github repo -- however, I seem to be unable to run SDRPlusPlus after a successful compile on Ubuntu 20.10 (amd64):

```
$ ./sdrpp -r ../root_dev/
[2021-02-17 12:14:57.705] [info] SDR++ v0.2.5_beta
[2021-02-17 12:14:57.705] [info] Loading config
[2021-02-17 12:14:57.705] [warning] Config file '../root_dev//config.json' does not exist, creating it
sdrpp: /home/ubuntu/builds/SDRPlusPlus/core/src/imgui/stb_image_resize.h:2394: int stbir__resize_allocated(stbir__info*, const void*, int, void*, int, int, stbir_uint32, stbir_datatype, stbir_edge, stbir_edge, stbir_colorspace, void*, size_t): Assertion `(size_t)(unsigned char*)(((unsigned char*)info->ring_buffer) + info->ring_buffer_size) == (size_t)tempmem + tempmem_size_in_bytes' failed.
Aborted (core dumped)
```

If I run `sdrpp` alone, I get the same error, but without the config file warning:

```
$ ./sdrpp 
[2021-02-17 12:15:54.548] [info] SDR++ v0.2.5_beta
[2021-02-17 12:15:54.548] [info] Loading config
sdrpp: /home/ubuntu/builds/SDRPlusPlus/core/src/imgui/stb_image_resize.h:2394: int stbir__resize_allocated(stbir__info*, const void*, int, void*, int, int, stbir_uint32, stbir_datatype, stbir_edge, stbir_edge, stbir_colorspace, void*, size_t): Assertion `(size_t)(unsigned char*)(((unsigned char*)info->ring_buffer) + info->ring_buffer_size) == (size_t)tempmem + tempmem_size_in_bytes' failed.
Aborted (core dumped)
```

Running from the main directory (one directory level up, sdrpp has been built under 'build') produces this:

```
$ ./build/sdrpp -r root_dev
[2021-02-17 12:18:40.935] [info] SDR++ v0.2.5_beta
[2021-02-17 12:18:40.936] [info] Loading config
sdrpp: /home/ubuntu/builds/SDRPlusPlus/core/src/imgui/stb_image_resize.h:1141: void stbir__normalize_downsample_coefficients(stbir__contributors*, float*, stbir_filter, float, int, int): Assertion `total > 0.9f' failed.
Aborted (core dumped)```

All help appreciated.

AlexandreRouma, 2021-02-17:
Hello,
when running from the build directory, you need to follow the instructions in the readme on creating and configuring a root directory

nonoo, 2021-03-15:
I also have this issue on Lubuntu 20.10. I have the following module definitions in config.json:

```
    "moduleInstances": {
        "Airspy Source": "airspy_source",
        "AirspyHF+ Source": "airspyhf_source",
        "Audio Sink": "audio_sink",
        "PlutoSDR Source": "plutosdr_source",
        "RTL-SDR Source": "rtl_sdr_source",
        "RTL-TCP Source": "rtl_tcp_source",
        "Radio": "radio",
        "Recorder": "recorder",
        "SoapySDR Source": "soapy_source"
    },
    "modules": [
        "./radio/radio.so",
        "./recorder/recorder.so",
        "./file_source/file_source.so",
        "./rtl_tcp_source/rtl_tcp_source.so",
        "./soapy_source/soapy_source.so",
        "./airspyhf_source/airspyhf_source.so",
        "./airspy_source/airspy_source.so",
        "./plutosdr_source/plutosdr_source.so",
        "./rtl_sdr_source/rtl_sdr_source.so",
        "./audio_sink/audio_sink.so"
    ],
    "modulesDirectory": "./root_dev/modules",
    "offset": 0.0,
    "resourcesDirectory": "./root_dev/res",
```

I have ran the create_root.sh script and also added the modules listing using the find command described in the README.

gdb backtrace:

```
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:49
#1  0x00007ffff7724864 in __GI_abort () at abort.c:79
#2  0x00007ffff7724749 in __assert_fail_base (fmt=0x7ffff78b0458 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", 
    assertion=0x7ffff7b6756e "total > 0.9f", file=0x7ffff7b66588 "/usr/local/src/SDRPlusPlus/core/src/imgui/stb_image_resize.h", 
    line=1141, function=<optimized out>) at assert.c:92
#3  0x00007ffff7736a96 in __GI___assert_fail (assertion=0x7ffff7b6756e "total > 0.9f", 
    file=0x7ffff7b66588 "/usr/local/src/SDRPlusPlus/core/src/imgui/stb_image_resize.h", line=1141, 
    function=0x7ffff7b66c38 "void stbir__normalize_downsample_coefficients(stbir__contributors*, float*, stbir_filter, float, int, int)") at assert.c:101
#4  0x00007ffff79d129a in stbir__calculate_filters(stbir__contributors*, float*, stbir_filter, float, float, int, int) ()
   from /usr/local/src/SDRPlusPlus/build/core/libsdrpp_core.so
#5  0x00007ffff79d1baf in stbir__resize_arbitrary(void*, void const*, int, int, int, void*, int, int, int, float, float, float, float, float*, int, int, unsigned int, stbir_datatype, stbir_filter, stbir_filter, stbir_edge, stbir_edge, stbir_colorspace) [clone .constprop.0] () from /usr/local/src/SDRPlusPlus/build/core/libsdrpp_core.so
#6  0x00007ffff79d3d40 in stbir_resize_uint8 () from /usr/local/src/SDRPlusPlus/build/core/libsdrpp_core.so
#7  0x00007ffff79d652c in sdrpp_main(int, char**) () from /usr/local/src/SDRPlusPlus/build/core/libsdrpp_core.so
#8  0x00007ffff7726cb2 in __libc_start_main (main=0x555555555060 <main>, argc=3, argv=0x7fffffffd9f8, init=<optimized out>, 
    fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffd9e8) at ../csu/libc-start.c:314
#9  0x000055555555509e in _start ()
```

The following assert fails (stbir__normalize_downsample_coefficients() core/src/imgui/stb_image_resize.h:1141): STBIR_ASSERT(total > 0.9f);

nonoo, 2021-03-22:
I'm running SDR++ using -r, and it is pointing to the root directory created with create_root.sh, but you were right, it runs without issues after installing it!

alphafox02, 2021-04-11:
@nonoo in your last post you mentioned the error goes away after installing it. Just curious, what constitutes installing it? The readme for Linux has a “coming soon” to the install section. Just curious what more is required when not wanting to just have it setup for development. Thanks!

nonoo, 2021-04-11:
sudo make install :)

alphafox02, 2021-04-11:
Too easy, I was thinking there was maybe something else otherwise I couldn’t understand why the install section of Linux said “coming soon”.. I’ll see if that clears up my problem.

AlexandreRouma, 2021-04-12:
yeah should probably update that

closed 2021-04-12 by AlexandreRouma