AlexandreRouma/SDRPlusPlus #77: Building on Mac

ThijsCas, 2021-02-23:
I'm trying to build SDR++ on a mac using the "Building on Linux / BSD" section.

I think I installed all the required dependencies correctly, except for `libairspyhf`, `libiio`, `libad9361` and `librtaudio-dev`. I did install `SoapySDR`, which I think should be enough, but I do still get this error:

```
-- Checking for module 'libairspyhf'
--   No package 'libairspyhf' found
CMake Error at /usr/local/Cellar/cmake/3.19.5/share/cmake/Modules/FindPkgConfig.cmake:553 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.19.5/share/cmake/Modules/FindPkgConfig.cmake:741 (_pkg_check_modules_internal)
  airspyhf_source/CMakeLists.txt:26 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
```

Do I absolutely need to install `libairspyhf` and others, even if I don't want to support Airspy HF+?

aosync, 2021-02-23:
You can create the build files with `cmake .. -DOPT_BUILD_AIRSPYHF_SOURCE=OFF` to disable airspyhf support and dependency during the build. You can check `CMakeLists.txt` to get a list of build options for the different dependencies.

AlexandreRouma, 2021-02-23:
You need to install libairspyhf if you want to use it.
Otherwise, disable building that module as suggested above.

ThijsCas, 2021-02-23:
Thanks! After disabling the modules that I didn't want to install, I was able to correctly run cmake.

I still ran into an error trying to actually compile it, but I'm chalking that up to Mac not being properly supported yet, and I don't think I'll be able to fix that myself, so I'll let smarter people do that.

This is the start of the errors while compiling if anyone is interested.

```
Undefined symbols for architecture x86_64:
  "__DefaultRuneLocale", referenced from:
```

closed 2021-02-23 by ThijsCas

AlexandreRouma, 2021-02-23:
SDR++ should fully support OSX, just not for installing atm.
Make sure you compile with GCC btw.