AlexandreRouma/SDRPlusPlus #79: some issues

sdaviet, 2021-03-13:
Hello,
I don't have your mail. I use this page.
I'm working on some issues (from the last release because this version work on windows) like :
  - wav audio files sound level,
  - adding audio balance slider,

I do a first test with IQ file player.
I think memorize menu and windows state for the next startup.
It's possible to contribuate a little bit on your nice project.

have a good day

AlexandreRouma, 2021-03-14:
Hello,
the wav audio sound level and config were fixed in the new version of the recorder and audio sink.
An audio balances sliders might still be needed tho.

It has to be an addition to the Volume DSP block.

AlexandreRouma, 2021-05-09:
Closing since these features are implemented.

closed 2021-05-09 by AlexandreRouma