AlexandreRouma/SDRPlusPlus #81: need help for compile on macos

BG7JAF, 2021-03-21:
Hello, I try to compile this program on macos. compiling is ok, but when linking error occus. Here I upload the error log, how can I fix this error? thx

platform: macos 10.15.7

[log.txt](https://github.com/AlexandreRouma/SDRPlusPlus/files/6176855/log.txt)

AlexandreRouma, 2021-03-21:
Hello, clang is currently not supported.
Please use GCC instead.
MacOS support is general is a bit wonky atm since I don't have a VM to test it.

kd5ftn, 2021-03-22:
> MacOS support is general is a bit wonky atm since I don't have a VM to test it.

@AlexandreRouma You might be interested in https://snapcraft.io/sosumi

AlexandreRouma, 2021-03-22:
> 
> 
> > MacOS support is general is a bit wonky atm since I don't have a VM to test it.
> 
> @AlexandreRouma You might be interested in https://snapcraft.io/sosumi

Unfortunately, my main PC runs windows and my linux PC isn't fast enough to run a MacOS VM :/

closed 2021-03-23 by BG7JAF