AlexandreRouma/SDRPlusPlus #82: linux compile error

webislands, 2021-03-21:
is my firts compile sdrplusplus i am under linux sorry my english i am french ...

is my log need your help

└─$ cmake ..
-- Checking for module 'libairspyhf'
--   Found libairspyhf, version 1.6
-- Checking for module 'libairspy'
--   Found libairspy, version 1.0
-- Configuring done
-- Generating done
-- Build files have been written to: /home/webs/SDRPlusPlus/build
                                                                                                                                                                       
┌──(webs㉿node-1w7jra91wlivqcczazcsm3zht)-[~/SDRPlusPlus/build]
└─$ make -j$(nproc)                                        

[ 51%] Built target sdrpp_core
[ 55%] Built target sdrpp
[ 55%] Built target radio
[ 60%] Built target recorder
[ 60%] Built target file_source
[ 63%] Built target rtl_tcp_source
[ 64%] Building C object falcon9_decoder/CMakeFiles/falcon9_decoder.dir/src/libcorrect/convolutional/sse/decode.c.o
Scanning dependencies of target soapy_source
[ 65%] Building CXX object soapy_source/CMakeFiles/soapy_source.dir/src/main.cpp.o
In file included from /usr/lib/gcc/x86_64-linux-gnu/10/include/immintrin.h:35,
                 from /usr/lib/gcc/x86_64-linux-gnu/10/include/x86intrin.h:32,
                 from /home/webs/SDRPlusPlus/falcon9_decoder/src/libcorrect/correct/convolutional/sse/lookup.h:5,
                 from /home/webs/SDRPlusPlus/falcon9_decoder/src/libcorrect/correct/convolutional/sse/convolutional.h:2,
                 from /home/webs/SDRPlusPlus/falcon9_decoder/src/libcorrect/convolutional/sse/decode.c:1:
/home/webs/SDRPlusPlus/falcon9_decoder/src/libcorrect/convolutional/sse/decode.c: In function ‘convolutional_sse_decode_inner’:
/usr/lib/gcc/x86_64-linux-gnu/10/include/tmmintrin.h:136:1: error: inlining failed in call to ‘always_inline’ ‘_mm_shuffle_epi8’: target specific option mismatch
  136 | _mm_shuffle_epi8 (__m128i __X, __m128i __Y)
      | ^~~~~~~~~~~~~~~~
/home/webs/SDRPlusPlus/falcon9_decoder/src/libcorrect/convolutional/sse/decode.c:211:25: note: called from here
  211 |                 hist2 = _mm_shuffle_epi8(hist2, hist_mask);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/lib/gcc/x86_64-linux-gnu/10/include/immintrin.h:35,
                 from /usr/lib/gcc/x86_64-linux-gnu/10/include/x86intrin.h:32,
                 from /home/webs/SDRPlusPlus/falcon9_decoder/src/libcorrect/correct/convolutional/sse/lookup.h:5,
                 from /home/webs/SDRPlusPlus/falcon9_decoder/src/libcorrect/correct/convolutional/sse/convolutional.h:2,
                 from /home/webs/SDRPlusPlus/falcon9_decoder/src/libcorrect/convolutional/sse/decode.c:1:
/usr/lib/gcc/x86_64-linux-gnu/10/include/tmmintrin.h:136:1: error: inlining failed in call to ‘always_inline’ ‘_mm_shuffle_epi8’: target specific option mismatch
  136 | _mm_shuffle_epi8 (__m128i __X, __m128i __Y)
      | ^~~~~~~~~~~~~~~~
/home/webs/SDRPlusPlus/falcon9_decoder/src/libcorrect/convolutional/sse/decode.c:208:25: note: called from here
  208 |                 hist1 = _mm_shuffle_epi8(hist1, hist_mask);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/lib/gcc/x86_64-linux-gnu/10/include/immintrin.h:35,
                 from /usr/lib/gcc/x86_64-linux-gnu/10/include/x86intrin.h:32,
                 from /home/webs/SDRPlusPlus/falcon9_decoder/src/libcorrect/correct/convolutional/sse/lookup.h:5,
                 from /home/webs/SDRPlusPlus/falcon9_decoder/src/libcorrect/correct/convolutional/sse/convolutional.h:2,
                 from /home/webs/SDRPlusPlus/falcon9_decoder/src/libcorrect/convolutional/sse/decode.c:1:
/usr/lib/gcc/x86_64-linux-gnu/10/include/tmmintrin.h:136:1: error: inlining failed in call to ‘always_inline’ ‘_mm_shuffle_epi8’: target specific option mismatch
  136 | _mm_shuffle_epi8 (__m128i __X, __m128i __Y)
      | ^~~~~~~~~~~~~~~~
/home/webs/SDRPlusPlus/falcon9_decoder/src/libcorrect/convolutional/sse/decode.c:205:25: note: called from here
  205 |                 hist0 = _mm_shuffle_epi8(hist0, hist_mask);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/lib/gcc/x86_64-linux-gnu/10/include/immintrin.h:35,
                 from /usr/lib/gcc/x86_64-linux-gnu/10/include/x86intrin.h:32,
                 from /home/webs/SDRPlusPlus/falcon9_decoder/src/libcorrect/correct/convolutional/sse/lookup.h:5,
                 from /home/webs/SDRPlusPlus/falcon9_decoder/src/libcorrect/correct/convolutional/sse/convolutional.h:2,
                 from /home/webs/SDRPlusPlus/falcon9_decoder/src/libcorrect/convolutional/sse/decode.c:1:
/usr/lib/gcc/x86_64-linux-gnu/10/include/tmmintrin.h:136:1: error: inlining failed in call to ‘always_inline’ ‘_mm_shuffle_epi8’: target specific option mismatch
  136 | _mm_shuffle_epi8 (__m128i __X, __m128i __Y)
      | ^~~~~~~~~~~~~~~~
/home/webs/SDRPlusPlus/falcon9_decoder/src/libcorrect/convolutional/sse/decode.c:202:24: note: called from here
  202 |                 hist = _mm_shuffle_epi8(hist, hist_mask);
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... 289 lines cut ...]
                 from /home/webs/SDRPlusPlus/falcon9_decoder/src/libcorrect/correct/convolutional/sse/convolutional.h:2,
                 from /home/webs/SDRPlusPlus/falcon9_decoder/src/libcorrect/convolutional/sse/decode.c:1:
/usr/lib/gcc/x86_64-linux-gnu/10/include/tmmintrin.h:136:1: error: inlining failed in call to ‘always_inline’ ‘_mm_shuffle_epi8’: target specific option mismatch
  136 | _mm_shuffle_epi8 (__m128i __X, __m128i __Y)
      | ^~~~~~~~~~~~~~~~
/home/webs/SDRPlusPlus/falcon9_decoder/src/libcorrect/convolutional/sse/decode.c:208:25: note: called from here
  208 |                 hist1 = _mm_shuffle_epi8(hist1, hist_mask);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/lib/gcc/x86_64-linux-gnu/10/include/immintrin.h:35,
                 from /usr/lib/gcc/x86_64-linux-gnu/10/include/x86intrin.h:32,
                 from /home/webs/SDRPlusPlus/falcon9_decoder/src/libcorrect/correct/convolutional/sse/lookup.h:5,
                 from /home/webs/SDRPlusPlus/falcon9_decoder/src/libcorrect/correct/convolutional/sse/convolutional.h:2,
                 from /home/webs/SDRPlusPlus/falcon9_decoder/src/libcorrect/convolutional/sse/decode.c:1:
/usr/lib/gcc/x86_64-linux-gnu/10/include/tmmintrin.h:136:1: error: inlining failed in call to ‘always_inline’ ‘_mm_shuffle_epi8’: target specific option mismatch
  136 | _mm_shuffle_epi8 (__m128i __X, __m128i __Y)
      | ^~~~~~~~~~~~~~~~
/home/webs/SDRPlusPlus/falcon9_decoder/src/libcorrect/convolutional/sse/decode.c:211:25: note: called from here
  211 |                 hist2 = _mm_shuffle_epi8(hist2, hist_mask);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [falcon9_decoder/CMakeFiles/falcon9_decoder.dir/build.make:199 : falcon9_decoder/CMakeFiles/falcon9_decoder.dir/src/libcorrect/convolutional/sse/decode.c.o] Erreur 1
make[1]: *** [CMakeFiles/Makefile2:539 : falcon9_decoder/CMakeFiles/falcon9_decoder.dir/all] Erreur 2
make[1]: *** Attente des tâches non terminées....
[ 66%] Linking CXX shared library soapy_source.so
[ 66%] Built target soapy_source
make: *** [Makefile:149 : all] Erreur 2

AlexandreRouma, 2021-03-21:
hello, this is a bug from yesterday added by accident.
I'll post an update here once it's fixed

webislands, 2021-03-21:
tnk compil is work just lauch i have this msg
sdrpp: error while loading shared libraries: libvolk.so.1.4: cannot open shared object file: No such file or directory
do you have solution ???

AlexandreRouma, 2021-03-21:
this is a SoapySDR bug and isn't related to SDR++
From other users that have had this issues, it appears that it comes from the soapysdr-modules-all (or soapysdr-all) packages.
I recommend installing only the base SoapySDR and then installing only the soapy modules you need.

webislands, 2021-03-22:
sorry i am compile in another machine and distro i have it

[ 87%] Building CXX object airspy_source/CMakeFiles/airspy_source.dir/src/main.cpp.o
/home/webs/SDRPlusPlus/airspy_source/src/main.cpp: Dans la fonction membre « void AirspySourceModule::refresh() »:
/home/webs/SDRPlusPlus/airspy_source/src/main.cpp:77:17: erreur: « airspy_list_devices » n'a pas été déclaré dans cette portée; vouliez-vous employer « airspy_device » ?
   77 |         int n = airspy_list_devices(serials, 256);
      |                 ^~~~~~~~~~~~~~~~~~~
      |                 airspy_device
[ 89%] Linking CXX shared library airspyhf_source.so
[ 89%] Built target airspyhf_source
Scanning dependencies of target plutosdr_source
[ 90%] Building CXX object plutosdr_source/CMakeFiles/plutosdr_source.dir/src/main.cpp.o
make[2]: *** [airspy_source/CMakeFiles/airspy_source.dir/build.make:82 : airspy_source/CMakeFiles/airspy_source.dir/src/main.cpp.o] Erreur 1
make[1]: *** [CMakeFiles/Makefile2:599 : airspy_source/CMakeFiles/airspy_source.dir/all] Erreur 2
make[1]: *** Attente des tâches non terminées....
[ 92%] Linking CXX shared library plutosdr_source.so
[ 92%] Built target plutosdr_source
make: *** [Makefile:149 : all] Erreur 2

AlexandreRouma, 2021-03-22:
you have an outdated version of libairspy

webislands, 2021-03-24:
sorry lol do you have solution ...
<
[2021-03-24 18:24:57.339] [info] Initializing RTL-SDR Source (rtl_sdr_source)
Detached kernel driver
Found Rafael Micro R820T tuner
r82xx_write: i2c wr failed=-1 reg=0b len=1
r82xx_init: failed=-1
rtlsdr_demod_write_reg failed with -1
rtlsdr_demod_read_reg failed with -1
rtlsdr_demod_write_reg failed with -1
rtlsdr_demod_read_reg failed with -1
rtlsdr_demod_write_reg failed with -1
rtlsdr_demod_read_reg failed with -1
rtlsdr_write_reg failed with -1
Reattaching kernel driver failed!
[2021-03-24 18:24:57.718] [info] New DSP samplerate: 250000.0
[2021-03-24 18:24:57.718] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2021-03-24 18:24:57.720] [info] Initializing Radio (radio)
[2021-03-24 18:24:57.764] [info] Initializing Recorder (recorder)
[2021-03-24 18:24:57.765] [error] Cannot unbind from stream ''. Stream doesn't exist
[2021-03-24 18:24:57.765] [info] Initializing SoapySDR Source (soapy_source)
[INFO] [UHD] linux; GNU C++ version 10.2.1 20201207; Boost_107400; UHD_3.15.0.0-4+b1
[ERROR] avahi_service_browser_new() failed: Bad state
Hash collision!!! Fatal error!!

AlexandreRouma, 2021-03-25:
Likely some permissions issue

AlexandreRouma, 2021-04-30:
Closing for inactivity

closed 2021-04-30 by AlexandreRouma