AlexandreRouma/SDRPlusPlus #87: libGLEW dependency problem

amir1376irani, 2021-03-28:
Hi
i'm running ubuntu 18.04 with kernel 5.4 on intel core-i3 350m without dedicated graphic card
downloaded your provided [.deb file release v0.2.5 _beta PREVIEW #3](https://github.com/AlexandreRouma/SDRPlusPlus/releases/download/0.2.5_beta_preview3/sdrpp_debian_amd64.deb)
followed your [install guide](https://github.com/AlexandreRouma/SDRPlusPlus#linux)
when i run `sdrpp` it says `sdrpp: error while loading shared libraries: libGLEW.so.2.1: cannot open shared object file: No such file or directory`
checked my libraries ans saw libGLEW2.0 is installed (shipped with ubuntu 18.04)
then compiled [glew repository v2.1.0](https://github.com/nigels-com/glew/releases/download/glew-2.1.0/glew-2.1.0.zip) [although newer version 2.2 is available but  i compiled version 2.1 because sdrpp said it requires version 2.1]
glew makefile by default will takes the compiled .so file into `/usr/lib64` so added that temporary with `export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib64`
now when i run `sdrpp` will get this error:
```
[2021-03-28 20:08:30.178] [info] SDR++ v0.2.5_beta
[2021-03-28 20:08:30.179] [info] Loading config
[2021-03-28 20:08:30.229] [error] Glfw Error 65543: GLX: Failed to create context: GLXBadFBConfig
```
after searching about this problem i didn't get what's wrong or incompatible, and here is my `glxinfo |grep "OpenGL version"` output : `OpenGL version string: 2.1 Mesa 20.0.8`

Any help appreciated!

AlexandreRouma, 2021-03-28:
Hello,

This build requires OpenGL 3.0, your PC seems to only support OpenGL 2.1

You could try to edit lines 184 though 186 to allow older OpenGL versions, but I doubt your PC will have the performance required to run SDR software.

PS: I don't have hardware old enough to test this fix.

amir1376irani, 2021-03-28:
tnx, i will test and make report

AlexandreRouma, 2021-04-12:
Hello, any update?

AlexandreRouma, 2021-04-30:
Closing for inactivity, reopen if the issue persists

closed 2021-04-30 by AlexandreRouma