AlexandreRouma/SDRPlusPlus #89: SDR++ keeps running for a few seconds after closing window

shuyuan-liu, 2021-03-30:
I'm using an RTL-SDR and built SDR++ on Linux. Everything works fine, but when I close SDR++ by clicking the window's close button, the window closes but audio keeps coming through for a few seconds before stopping. looking in `htop` I can confirm that the `sdrpp` process runs for a further 5 to 10 seconds before stopping.

### Build info

- gcc 10.2.0
- fftw 3.3.9
- glfw 3.3.3
- glew 2.2.0
- volk 2.4.1
- soapysdr 0.7.2
- rtaudio 5.1.0
- SDR++ commit bcc1744

### System info

Arch Linux 5.11.10-arch1-1, using PulseAudio and GNOME on X11.

shuyuan-liu, 2021-03-30:
When the process actually exits, it prints the following in the terminal:

```
terminate called without an active exception
Job 1, 'sdrpp' terminated by signal SIGABRT (Abort)
```

No other errors are printed.

shuyuan-liu, 2021-03-30:
This seems unrelated to the type of SDR used, as the problem still appears when opening then closing SDR++ with other inputs selected, e.g. Airspy.

Below is a backtrace from gdb when the crash happens:

```
#0  0x00007ffff67c9ef5 in raise () from /usr/lib/libc.so.6
#1  0x00007ffff67b3862 in abort () from /usr/lib/libc.so.6
#2  0x00007ffff6b5186a in __gnu_cxx::__verbose_terminate_handler () at /build/gcc/src/gcc/libstdc++-v3/libsupc++/vterminate.cc:95
#3  0x00007ffff6b5dd3a in __cxxabiv1::__terminate (handler=<optimized out>) at /build/gcc/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:48
#4  0x00007ffff6b5dda7 in std::terminate () at /build/gcc/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:58
#5  0x00007ffff7b45f26 in dsp::Reshaper<dsp::complex_t>::~Reshaper() () from /home/shuyuan/Build/SDRPlusPlus/build/core/libsdrpp_core.so
#6  0x00007ffff7b4ab62 in SignalPath::~SignalPath() () from /home/shuyuan/Build/SDRPlusPlus/build/core/libsdrpp_core.so
#7  0x00007ffff67ccc8e in __cxa_finalize () from /usr/lib/libc.so.6
#8  0x00007ffff796db68 in __do_global_dtors_aux () from /home/shuyuan/Build/SDRPlusPlus/build/core/libsdrpp_core.so
#9  0x00007fffffffe140 in ?? ()
#10 0x00007ffff7fdc23b in _dl_fini () from /lib64/ld-linux-x86-64.so.2
```

It is interesting to note that after closing the window, gdb shows thread exiting one by one and one per second (roughly). In other words, the following is printed in gdb, one line per second, after closing the window:

```
[Thread 0x7fffe2ffd640 (LWP 26731) exited]
[Thread 0x7fffe37fe640 (LWP 26730) exited]
[Thread 0x7fffe3fff640 (LWP 26729) exited]
[Thread 0x7fffe8f23640 (LWP 26728) exited]
[Thread 0x7fffc2ffd640 (LWP 26744) exited]
[Thread 0x7fffc37fe640 (LWP 26743) exited]
[Thread 0x7fffc3fff640 (LWP 26742) exited]
...
```

It seems too slow on an i7-8565U system for threads to exit like this, so there might be something wrong with that. Not sure if it's related to the SIGABRT though.

AlexandreRouma, 2021-03-31:
Hello,
this is a known issue due to certain threads not being terminated before the software exits.

closed 2021-04-08 by shuyuan-liu