AlexandreRouma/SDRPlusPlus #92: Large recording files have broken length - maybe use different format?

shuyuan-liu, 2021-04-02:
I made a recording this morning at a sample rate of 1.024 M for about half and hour, resulting in a file size of 5.21 GB. However when playing back this recording, most software I tried only recognised and showed / processed the first 5 minutes or so of it, disregarding the rest, including GNU Radio, SDR#, meteor_demod, and Audacity.

I think standard wav files can only go up to 4 GiB according to [Wikipedia](https://en.wikipedia.org/wiki/WAV). There is the RF64 format that uses a different header and allows much larger files, but I don't know if RF64 is compatible with the programs mentioned above.

Some possible ways I can think of to solve this problem:
- Use the headerless 32-bit float .raw format from GNU Radio, or a custom 16-bit / 8-bit int format
- Automatically create and record into another file every time size reaches 4 GiB
- Use RF64

AlexandreRouma, 2021-04-12:
Wav64 support is planned. haven't added it yet as recording and playing back files over 4gb is still supported by SDR++

AlexandreRouma, 2023-02-22:
New recorder with planned support for RF64 has been added a few weeks ago but it'll take a bit of time to support RF64 cleanly. I don't want to hack anything in