AlexandreRouma/SDRPlusPlus #93: Feature Request: CTCSS / DCS tone decoding.

ellisgeek, 2021-04-07:
At some point in the future™ it would be really helpful if the ability to decode CTCSS and DCS tones could be added. A) to locate the tone being used on a frequency and B) to squelch the audio output using the decoded tone.

This was available in SDR# via 2 plugins from rtl-sdr.ru.

P.S. SDR++ is incredibly impressive!

AlexandreRouma, 2021-04-12:
Will keep in mind, not sure how hard it is to implement.

rough316, 2021-04-26:
I would really like this CTCSS / PL tone decoding squelch as well. Trying to use the SDR on a crowded channel and this would be very helpful.

hicaoc, 2021-08-04:
Also need to implement filtering, for CTCSS, if there is CTCSS in the signal, then the audio output needs to be this dumb audio filter out, otherwise the horn or headphones will have this low frequency, affecting the hearing.

kieths, 2021-08-18:
I just installed and tested SDR++ today.  I was looking for this same feature (CTCSS).  I see that SDRSharp has a plugin for this ([link](http://rtl-sdr.ru/page/ctcss-detektor-shumopodavitel)), as well as other plugins.  I'm not see how to add plugins to SDR++ (?).

AlexandreRouma, 2021-08-18:
SDR# plugins aren't compatible with SDR++

craigerl, 2021-08-22:
Agreed, it's hard to use this with repeater systems which transmit a PL tone.

You can use sox to filter it out, but really inconvenient, and introduces some lag.

socat stdout udp-listen:7355 | play   -t raw -b 16 -e signed  -r 48k - -r 16k sinc -n 32767  180-8000

AlexandreRouma, 2023-03-27:
Note that an optional audio high-pass has been added which can get rid of the PL tones.
Still can't squelch or decode it though.

NazimkovMykola, 2025-01-27:
I understand this option is still not implemented?

craigerl, 2025-01-27:
It is implemented - check "High pass filter", PL tone goes away, works great.

agausmann, 2025-01-27:
> It is implemented - check "High pass filter", PL tone goes away, works great.

This is not the same as decoding the tone or using it to open squelch.