AlexandreRouma/SDRPlusPlus #98: [Feature] frequency control with keyboard

shuyuan-liu, 2021-04-19:
It seems currently in SDR++ the frequency can only be adjusted with the mouse, by either clicking on the digits or hovering over them and scrolling. However, in some scenarios it would be easier to control the frequency via the keyboard, for example when using SDR++ on a laptop out in the fields. There are two methods of frequency control I can think of:

- Entering a literal frequency value in a textbox, maybe with a selectable unit kHz, MHz, etc.
- Adjusting the frequency's digits with the arrow keys. Left and right moves the cursor, up and down increase and decrease the digit under the cursor, like in GQRX and SDR#.

AlexandreRouma, 2021-04-19:
Good idea!
I'll keep it in mind.

AlexandreRouma, 2021-04-23:
This is now implemented:
You can type in the frequency by hovering the first digit you want to modify, then just type the numbers

To use the arrow keys on the frequency select, hover over a digit, use left/right to go to another digit, use up down to increase or decrease that digit.

Finally, if you use the arrow keys outside of the frequency select, the VFO will move left or right by the step size of that VFO

closed 2021-04-23 by AlexandreRouma