AlexandreRouma/SDRPlusPlus #99: LINK : fatal error LNK1181: cannot open input file 'rtaudio.lib'

XxOinvizioNxX, 2021-04-19:
I am trying to build SDR++ on Windows. Installed all dependencies, including the rtaudio build.
In PATH I have:
- C:\Program Files\RtAudio
- C:\Program Files\RtAudio\bin
- C:\Program Files\RtAudio\include
- C:\Program Files\RtAudio\lib
- C:\Program Files\RtAudio\share

These directories contain rtaudio.dll, RtAudio.h, rtaudio.lib, etc.

But when I try to build the SDR++, the error  `LNK1181: cannot open input file 'rtaudio.lib' ` occurs

Full log:
```
1>------ Build started: Project: sdrpp_core, Configuration: Release x64 ------
1>Auto build dll exports
1>LINK : fatal error LNK1181: cannot open input file 'rtaudio.lib'
1>Done building project "sdrpp_core.vcxproj" -- FAILED.
2>------ Build started: Project: sdrpp, Configuration: Release x64 ------
3>------ Build started: Project: airspy_source, Configuration: Release x64 ------
4>------ Build started: Project: airspyhf_source, Configuration: Release x64 ------
5>------ Build started: Project: audio_sink, Configuration: Release x64 ------
6>------ Build started: Project: file_source, Configuration: Release x64 ------
7>------ Build started: Project: hackrf_source, Configuration: Release x64 ------
8>------ Build started: Project: meteor_demodulator, Configuration: Release x64 ------
9>------ Build started: Project: plutosdr_source, Configuration: Release x64 ------
10>------ Build started: Project: radio, Configuration: Release x64 ------
11>------ Build started: Project: recorder, Configuration: Release x64 ------
12>------ Build started: Project: rtl_sdr_source, Configuration: Release x64 ------
13>------ Build started: Project: rtl_tcp_source, Configuration: Release x64 ------
2>LINK : fatal error LNK1181: cannot open input file 'core\Release\sdrpp_core.lib'
3>LINK : fatal error LNK1181: cannot open input file '..\core\Release\sdrpp_core.lib'
6>LINK : fatal error LNK1181: cannot open input file '..\core\Release\sdrpp_core.lib'
4>LINK : fatal error LNK1181: cannot open input file '..\core\Release\sdrpp_core.lib'
7>LINK : fatal error LNK1181: cannot open input file '..\core\Release\sdrpp_core.lib'
2>Done building project "sdrpp.vcxproj" -- FAILED.
3>Done building project "airspy_source.vcxproj" -- FAILED.
14>------ Build started: Project: do_always, Configuration: Release x64 ------
6>Done building project "file_source.vcxproj" -- FAILED.
4>Done building project "airspyhf_source.vcxproj" -- FAILED.
7>Done building project "hackrf_source.vcxproj" -- FAILED.
10>LINK : fatal error LNK1181: cannot open input file '..\core\Release\sdrpp_core.lib'
10>Done building project "radio.vcxproj" -- FAILED.
9>LINK : fatal error LNK1181: cannot open input file '..\core\Release\sdrpp_core.lib'
9>Done building project "plutosdr_source.vcxproj" -- FAILED.
8>LINK : fatal error LNK1181: cannot open input file '..\core\Release\sdrpp_core.lib'
8>Done building project "meteor_demodulator.vcxproj" -- FAILED.
13>LINK : fatal error LNK1181: cannot open input file '..\core\Release\sdrpp_core.lib'
13>Done building project "rtl_tcp_source.vcxproj" -- FAILED.
11>LINK : fatal error LNK1181: cannot open input file '..\core\Release\sdrpp_core.lib'
11>Done building project "recorder.vcxproj" -- FAILED.
15>------ Build started: Project: do_always_volk, Configuration: Release x64 ------
16>------ Build started: Project: soapy_source, Configuration: Release x64 ------
17>------ Build started: Project: weather_sat_decoder, Configuration: Release x64 ------
12>LINK : fatal error LNK1181: cannot open input file '..\core\Release\sdrpp_core.lib'
12>Done building project "rtl_sdr_source.vcxproj" -- FAILED.
5>LINK : fatal error LNK1181: cannot open input file '..\core\Release\sdrpp_core.lib'
5>Done building project "audio_sink.vcxproj" -- FAILED.
14>File not found - *.dll
14>0 File(s) copied
15>C:/Program Files/PothosSDR/bin\volk.dll
15>1 File(s) copied
16>LINK : fatal error LNK1181: cannot open input file '..\core\Release\sdrpp_core.lib'
16>Done building project "soapy_source.vcxproj" -- FAILED.
17>LINK : fatal error LNK1181: cannot open input file '..\core\Release\sdrpp_core.lib'
17>Done building project "weather_sat_decoder.vcxproj" -- FAILED.
========== Build: 2 succeeded, 15 failed, 2 up-to-date, 0 skipped ==========
```

My steps:
- Clone this github repository
- `mkdir build`
- `cd build`
- `cmake .. "-DCMAKE_TOOLCHAIN_FILE=C:/Portables/vcpkg/scripts/buildsystems/vcpkg.cmake" -G "Visual Studio 16 2019"`
- Open .sln in Visual Studio
- Select "Release" configuration
- ALL_BUILD -> Build

XxOinvizioNxX, 2021-04-19:
[image]
[image]

XxOinvizioNxX, 2021-04-19:
I also tried installing rtaudio via vcpkg, but that didn't work

AlexandreRouma, 2021-04-19:
Hello,
Unfortunately, the current path for the RtAudio .lib is hardcoded to `C:/Program Files (x86)/RtAudio/lib` in audio_sink/CMakeLists.txt

As a quick fix, you could just modify that file but that would cuase issues if you want to contribute changes and pull the latest changes.

A fix would be to add a variable for it, I'll have to look into it.

XxOinvizioNxX, 2021-04-19:
> Hello,
> Unfortunately, the current path for the RtAudio .lib is hardcoded to `C:/Program Files (x86)/RtAudio/lib` in audio_sink/CMakeLists.txt
> 
> As a quick fix, you could just modify that file but that would cause issues if you want to contribute changes and pull the latest changes.
> 
> A fix would be to add a variable for it, I'll have to look into it.

Thank you so much! I just moved the `RtAudio` folder to `C:/Program Files (x86)/` and everything worked!

`========== Build: 17 succeeded, 0 failed, 2 up-to-date, 0 skipped ==========`

closed 2021-04-19 by XxOinvizioNxX