AlexandreRouma/SDRPlusPlus #100: Run-time error on Ubuntu 20.10

darrenfreeman, 2021-04-21:
I have followed the instructions to compile the latest git master from source. Commit 6332b33f3e5d6b6598c0664e0caadbec12b9293d.

This is what I get:

```
./build/sdrpp -r root_dev

[2021-04-21 14:18:56.857] [info] SDR++ v0.3.0_beta
[2021-04-21 14:18:56.857] [info] Loading config
sdrpp: /opt/SDRPlusPlus/core/src/imgui/stb_image_resize.h:1141: void stbir__normalize_downsample_coefficients(stbir__contributors*, float*, stbir_filter, float, int, int): Assertion `total > 0.9f' failed.
Aborted (core dumped)
```

I don't know how to begin investigating this. Please help :)

I have not yet tried a binary package, have never used the application before today.

AlexandreRouma, 2021-04-21:
Hello,

As explained [this part of the readme](https://github.com/AlexandreRouma/SDRPlusPlus#running-for-development), you must configure the paths to the resources when running without installing. Make sure that the paths you have configured are correct.

Regards,
Alex

darrenfreeman, 2021-04-22:
If that's indeed the problem, then there are missing error messages. An assert being triggered is a problem with the code, not the configuration.

darrenfreeman, 2021-04-22:
I have fixed a typo in the modules/resources directory names. Now I hit a different assertion in the same file.

```
./build/sdrpp -r root_dev

[2021-04-22 16:53:05.509] [info] SDR++ v0.3.0_beta
[2021-04-22 16:53:05.509] [info] Loading config
sdrpp: /opt/SDRPlusPlus/core/src/imgui/stb_image_resize.h:2394: int stbir__resize_allocated(stbir__info*, const void*, int, void*, int, int, stbir_uint32, stbir_datatype, stbir_edge, stbir_edge, stbir_colorspace, void*, size_t): Assertion `(size_t)(unsigned char*)(((unsigned char*)info->ring_buffer) + info->ring_buffer_size) == (size_t)tempmem + tempmem_size_in_bytes' failed.
Aborted (core dumped)
```

darrenfreeman, 2021-04-22:
OK, got it working. I was reading the example config for Windows, as that's the one you hyperlinked to. I didn't notice there are two examples, and they are slightly different.

So the real issue is the assertions being triggered when a file is not found.

AlexandreRouma, 2021-05-13:
Checks are now in place.

closed 2021-05-13 by AlexandreRouma