AlexandreRouma/SDRPlusPlus #103: Core dump on application close

darrenfreeman, 2021-04-22:
On Ubuntu 20.10, x86-64. When I closed the application after briefly playing with the UI, this is what I got. (Only the last few lines, of the console output, are shown.)

I haven't got any useful SoapySDR sources setup, so I didn't process any signals. Just open, poke around, close.

Compiled from source, from git master commit 6332b33f3e5d6b6598c0664e0caadbec12b9293d.

```
[2021-04-22 17:13:57.930] [info] SoapyModule 'SoapySDR Source': Menu Select!
[2021-04-22 17:13:57.930] [info] New DSP samplerate: 48000.0
[2021-04-22 17:13:57.937] [info] SoapyModule 'SoapySDR Source': Tune: 10024000.0!
[2021-04-22 17:14:21.584] [info] SoapyModule 'SoapySDR Source': Tune: 10000000.0!

terminate called without an active exception
Aborted (core dumped)
```

shuyuan-liu, 2021-04-22:
Seems to be the same issue as #89. Commit 4849d9c should have fixed the problem and building on my machine confirms that. Could you build and confirm @darrenfreeman?

darrenfreeman, 2021-04-22:
Seems fixed in 89e805b1a0dd187c31cd50e7e9c6a31a7cef4def, which is a bit newer still.

AlexandreRouma, 2021-04-22:
Thanks for confirming, closing the issue

closed 2021-04-22 by AlexandreRouma