AlexandreRouma/SDRPlusPlus #105: No audio output, core dump

darrenfreeman, 2021-04-22:
I am using an SDRPlay as the source, under Ubuntu 20.10.

I have so far not heard any sound output from SDR++, and it hasn't appeared in my PulseAudio sound control as an application that's producing sound. After changing the audio sink UI control between the various options, I got a core dump:

```
[2021-04-22 20:29:42.611] [info] SDRPlaySourceModule 'SDRplay Source': Tune: 7247000.0!
[2021-04-22 20:30:14.107] [info] RtAudio stream open
[2021-04-22 20:30:25.008] [info] RtAudio stream open
[2021-04-22 20:30:27.846] [info] RtAudio stream open
[2021-04-22 20:31:02.257] [info] RtAudio stream open
[2021-04-22 20:31:10.238] [error] Could not open audio device
terminate called after throwing an instance of 'RtAudioError'
  what():  RtApi:: a stream is not open!
Aborted (core dumped)
```

Here is another example.

```
[2021-04-22 20:34:26.067] [info] SDRPlaySourceModule 'SDRplay Source': Tune: 0.0!
[2021-04-22 20:34:26.505] [error] Could not open audio device
[2021-04-22 20:34:26.525] [info] SDRPlaySourceModule 'SDRplay Source': Tune: 7247000.0!
[2021-04-22 20:34:26.525] [info] Ready.
[2021-04-22 20:34:26.525] [info] SDRPlaySourceModule 'SDRplay Source': Tune: 7247000.0!
terminate called after throwing an instance of 'RtAudioError'
  what():  RtApi:: a stream is not open!
Aborted (core dumped)
```

AlexandreRouma, 2021-04-22:
hello,

Seems like a bad install of RtAudio. Please uninstall your current version and reinstall RtAudio from source.

darrenfreeman, 2021-04-22:
Uninstalled the rtaudio packages. Compiled the git master and installed. From the output of cmake,

```
-- Checking for module 'rtaudio'
--   Found rtaudio, version 5.1.0
```

Now, if I try to change the audio output, or even if I simply set the sink from Audio to None,

```
[2021-04-22 22:45:02.176] [info] RTLSDRSourceModule 'RTL-SDR Source': Tune: 0.0!
[2021-04-22 22:45:02.583] [warning] ConfigManager locked, waiting...
[2021-04-22 22:45:02.631] [error] Could not open audio device
[2021-04-22 22:45:02.654] [info] RTLSDRSourceModule 'RTL-SDR Source': Tune: 7150000000.0!
[2021-04-22 22:45:02.654] [info] Ready.
[2021-04-22 22:45:02.654] [info] RTLSDRSourceModule 'RTL-SDR Source': Tune: 7150000000.0!
terminate called after throwing an instance of 'RtAudioError'
  what():  RtApi:: a stream is not open!
Aborted (core dumped)
```

If I hit play without changing anything, the UI freezes and nothing appears to happen.

cropinghigh, 2021-04-23:
Maybe try to debug it with gdb?
gdb --args ./sdrpp -r ../root_dev
if you running it from build dir
To run, press r
Before it launched completely, press "Enter" when you see `--Type <RET> for more, q to quit, c to continue without paging--`
And enter "c" if you see "SIG32"
Then cause error to occur,
You will see same `--Type <RET> for more, q to quit, c to continue without paging--`
Press enter, then you can debug. 
Enter "bt" to get backtrace

AlexandreRouma, 2021-04-23:
RtAudio clearly can't open the device you're trying to select. This is likely a RtAudio issue or some other linux audio issue.

darrenfreeman, 2021-04-24:
It may be also a RtAudio issue, but SDR++ should never abort / core dump.

darrenfreeman, 2021-04-24:
```
/opt/rtaudio/build/tests$ ./audioprobe

Device Name = system
Device ID = 0
Probe Status = Successful
Output Channels = 2
Input Channels = 2
Duplex Channels = 2
This is the default output device.
This is the default input device.
Natively supported data formats:
  32-bit float
Supported sample rates = 48000 
Preferred sample rate = 48000
```

All of the detected devices look similar, and the bug is triggered if I select "system".

Note I also said it core dumps if I select Sink => None.

AlexandreRouma, 2021-04-24:
> Note I also said it core dumps if I select Sink => None.

Setting Sink to none still requires interaction with RtAudio to disable the audio stream.

AlexandreRouma, 2021-04-24:
I cannot replicate this issue on any system or VM I own. I'm gonna need for information to find the issue like the name of the devices available and running the rtaudio tests to figure out what framework it's trying to use.

darrenfreeman, 2021-04-25:
Could it not core dump please. If you add some error messages, it's possible I could fix it on my own without needing anyone to try to replicate.

darrenfreeman, 2021-04-25:
```
./apinames 

API names by identifier (C++):
* 4 'jack': 'Jack'
* 2 'pulse': 'Pulse'
* 1 'alsa': 'ALSA'
API identifiers by name (C++):
* 'jack': 4
* 'pulse': 2
* 'alsa': 1
API names by identifier (C):
* 4 'jack': 'Jack'
* 2 'pulse': 'Pulse'
* 1 'alsa': 'ALSA'
API identifiers by name (C):
* 'jack': 4
* 'pulse': 2
* 'alsa': 1
```

```
./audioprobe 

RtAudio Version 5.1.0

Compiled APIs:
  Jack Client
  Linux PulseAudio
  Linux ALSA

Current API: Jack Client

Found 7 device(s) ...

Device Name = system
Device ID = 0
Probe Status = Successful
Output Channels = 2
Input Channels = 2
Duplex Channels = 2
This is the default output device.
This is the default input device.
Natively supported data formats:
  32-bit float
Supported sample rates = 48000 
Preferred sample rate = 48000

Device Name = PulseAudio JACK Sink
Device ID = 1
Probe Status = Successful
Output Channels = 0
Input Channels = 2
Duplex Channels = 0
This is NOT the default output device.
This is NOT the default input device.
Natively supported data formats:
  32-bit float
Supported sample rates = 48000 
Preferred sample rate = 48000

Device Name = PulseAudio JACK Source
Device ID = 2
Probe Status = Successful
Output Channels = 2
Input Channels = 0
Duplex Channels = 0
This is NOT the default output device.
This is NOT the default input device.
Natively supported data formats:
  32-bit float
Supported sample rates = 48000 
Preferred sample rate = 48000

Device Name = GQRX
Device ID = 3
Probe Status = Successful
Output Channels = 0
Input Channels = 2
Duplex Channels = 0
This is NOT the default output device.
This is NOT the default input device.
Natively supported data formats:
  32-bit float
Supported sample rates = 48000 
Preferred sample rate = 48000

Device Name = WSJTX-In
Device ID = 4
Probe Status = Successful
Output Channels = 1
Input Channels = 0
Duplex Channels = 0
This is NOT the default output device.
This is NOT the default input device.
Natively supported data formats:
  32-bit float
Supported sample rates = 48000 
Preferred sample rate = 48000

Device Name = WSJTX-Out
Device ID = 5
Probe Status = Successful
Output Channels = 0
Input Channels = 1
Duplex Channels = 0
This is NOT the default output device.
This is NOT the default input device.
Natively supported data formats:
  32-bit float
Supported sample rates = 48000 
Preferred sample rate = 48000

Device Name = FLDIGI
Device ID = 6
Probe Status = Successful
Output Channels = 1
Input Channels = 0
Duplex Channels = 0
This is NOT the default output device.
This is NOT the default input device.
Natively supported data formats:
  32-bit float
Supported sample rates = 48000 
Preferred sample rate = 48000
```

darrenfreeman, 2021-04-25:
`./playraw 2 48000 $filename 0 0`

This results in a JACK source named `RtApiJack` with two channels. I have confirmed that sound is produced.

I am not seeing a new JACK source when I run SDR++. The audio devices on offer from within the application, are named after existing JACK sinks. I'm not sure what that is supposed to mean, is this going to create connections from the SDR++ JACK source to those sinks?

Also I never asked SDR++ to use JACK, it would be easier to just use PulseAudio for now. PulseAudio owns the physical sound device. But I love that it can support JACK eventually, as that's a more reliable way of piping sound into other applications.

closed 2022-09-13 by AlexandreRouma