AlexandreRouma/SDRPlusPlus #107: More de-emphasis options needed

JoshuaKimsey, 2021-04-24:
Firstly, let me say how amazing SDR++ is! I easily see this being able to compete with the likes of GQRX and Cubic on Linux. I can't wait for plugin support to be added so some SDR# plugins can be used! (Meteor demod would be so amazing!)

Now, for the issue I found. Normally in CubicSDR, when using the Stereo FM, I set de-emphasis to 25, as it provides a good middle ground between being a bit muffled and too tinny/noisy. Currently, the settings for SDR++ only has none, 50, and 75, which kinda presents a bit of an issue. If you could add in some more options, or the ability to set a custom value, either by hand or with a slider, it'd really help with regards to audio quality on the WFM, and even more so when stereo FM is added.

Thanks for the amazing work you're doing and I can't wait to see how SDR++ develops! I'm a full-time Linux user, so if you need any help testing out stuff for Linux, I'd be glad to offer my assistance! :)

AlexandreRouma, 2021-04-24:
Hello,

> I can't wait for plugin support to be added so some SDR# plugins can be used!

There is already plugin support. In fact, everything from each device to the radio is its own plugin. It's however impossible to add compatibility with SDR# for several reasons:

- Most SDR# plugins expect a windows environment
- SDR# plugins are written in C# which is extremely hard to interface from C++ (especially on linux/osx)
- SDR# plugins require the SDR# core libs which are not under any kind of open source license

> (Meteor demod would be so amazing!)

There is already a meteor demodulator plugin. It will soon be merged with the Weather Sat live decoder plugin that can decode down to image anything from LRPT (meteor) to to HRPT for NOAA and Meteor.

> Currently, the settings for SDR++ only has none, 50, and 75

There are international standard for preemphasis on WFM broadcast (75 for US and South Korea, 50 for the rest of the world). No other value is used. If you want a more muffled sound, you can sent the output sample rate to something like 12500Hz and you'll get a similar affect to the deemphasis filter. In the future there will also be audio noise reduction in the radio module. I think adding a custom option for the deemphasis would likely confuse users.

Best regards,
Alex

JoshuaKimsey, 2021-04-25:
> There is already plugin support. In fact, everything from each device to the radio is its own plugin.

Ah! Ok, that's also really awesome and should, in theory, make SDR++ very extensible for additions in the future. Like I said, I see it easily competing, and maybe beating, GQRX and CubicSDR on Linux. 

> It's however impossible to add compatibility with SDR#

I should have realized this. I forgot SDR# was written in C# (despite it literally being in the name of the program...), which would make anything written for it incompatible with SDR++. But that's ok! It allows for a fresh slate of plugins to be made for SDR++, which isn't a bad thing! Plus, C++ is a lot nicer to deal with on Linux than C#. C# is a pain...

> There is already a meteor demodulator plugin. It will soon be merged with the Weather Sat live decoder plugin

Ok, this made my heart jump for joy! As before my method for Receiving, Demoding, and Decoding LRPT involved no fewer than 4 steps, which was cumbersome and involved multiple different file types of stupid size... Being able to compress some of these steps together is gonna make this so much nicer to work with. Is APT expected to be added as well? Also, when will these plugins be unlocked for us to use in SDR++? I looked around but didn't see anyway to enable them, even though I see their folders in the main program folder.

> There are international standard for preemphasis on WFM broadcast (75 for US and South Korea, 50 for the rest of the world).

Ah, I didn't know these were international standards. In that case, it's not a huge priority for anything else. I just tended to use 25 as my preset in Cubic, which offers other options you see below: 
[image]

But, if you wanna just stick to international standards, then that's also fine by me too! Can't wait to see how this project evolves and also will look to see how I can help contribute to it. I'm not super well versed in C++, but I might be able to lend a hand here and there for smaller things! :)

closed 2021-04-28 by JoshuaKimsey