AlexandreRouma/SDRPlusPlus #109: spyserver_source: causes remote server to terminate on connection

10man, 2021-04-24:
I am new to this project and I will try to be of some use...

I have encountered a couple issues while trying to get my server working:
- The spyserver_source isn't enabled in the menu because it is not added to the config created on first run.
  - The config doesn't follow the modules that are available... It makes sense to me that the default config would be populated based on what modules were available during first run.
  - After manually updating the config file I have removed all warnings from the startup of `sdrpp`, I also see the spyserver source available in the source menu.
- When sdrpp connects to the remote server, the server terminate with the following output:
```shell
SPY Server v2.0.1700
Copyright (C) 2016-2018 Youssef Touil - https://airspy.com
Reading the configuration file: /home/ubuntu/spy-hf/spyserver.config
Listening for connections on 0.0.0.0:5555
Accepted client <private IP>:55324 running sdr++
Device was sleeping. Wake up!
Acquired an AirspyHF+ device
Caught signal 4
Terminating...
```
The last two lines repeat endlessly which is obviously a problem with the server! After a short time the following log is produced by the server:
```shell
Caught signal 4
Terminating...
Releasing the AirspyHF+ device
Closing all connections...
Caught signal 4
Terminating...
Caught signal 4
```

As for the logging from `sdrpp`, not much to report...
Here is the startup:
```shell
[2021-04-24 09:12:03.466] [info] SDR++ v0.3.0_beta
[2021-04-24 09:12:03.466] [info] Loading config
[2021-04-24 09:12:03.890] [info] Loading icons
[2021-04-24 09:12:03.901] [info] Loading band plans
[2021-04-24 09:12:03.905] [info] Loading band plans color table
[2021-04-24 09:12:03.916] [info] Loading modules
[2021-04-24 09:12:03.916] [info] Loading /usr/lib/sdrpp/plugins/radio.so
[2021-04-24 09:12:03.931] [info] Loading /usr/lib/sdrpp/plugins/airspyhf_source.so
[2021-04-24 09:12:03.946] [info] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[2021-04-24 09:12:03.962] [info] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[2021-04-24 09:12:03.979] [info] Loading /usr/lib/sdrpp/plugins/file_source.so
[2021-04-24 09:12:03.996] [info] Loading /usr/lib/sdrpp/plugins/recorder.so
[2021-04-24 09:12:04.015] [info] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[2021-04-24 09:12:04.035] [info] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so
[2021-04-24 09:12:04.048] [info] Loading /usr/lib/sdrpp/plugins/weather_sat_decoder.so
[2021-04-24 09:12:04.063] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2021-04-24 09:12:04.086] [info] New DSP samplerate: 768000.0
[2021-04-24 09:12:04.087] [info] Initializing Audio Sink (audio_sink)
[2021-04-24 09:12:04.094] [info] Initializing File Source (file_source)
[2021-04-24 09:12:04.113] [info] FileSourceModule 'File Source': Instance created!
[2021-04-24 09:12:04.113] [info] Initializing Radio (radio)
[2021-04-24 09:12:04.245] [info] Initializing Recorder (recorder)
[2021-04-24 09:12:04.246] [info] Initializing SpyServer Source (spyserver_source)
[2021-04-24 09:12:04.248] [info] Loading color maps
[2021-04-24 09:12:04.248] [info] Loading /usr/share/sdrpp/colormaps/websdr.json
[2021-04-24 09:12:04.249] [info] Loading /usr/share/sdrpp/colormaps/magma.json
[2021-04-24 09:12:04.250] [info] Loading /usr/share/sdrpp/colormaps/plasma.json
[2021-04-24 09:12:04.251] [info] Loading /usr/share/sdrpp/colormaps/electric.json
[2021-04-24 09:12:04.261] [info] Loading /usr/share/sdrpp/colormaps/turbo.json
[2021-04-24 09:12:04.278] [info] Loading /usr/share/sdrpp/colormaps/viridis.json
[2021-04-24 09:12:04.294] [info] Loading /usr/share/sdrpp/colormaps/inferno.json
[2021-04-24 09:12:04.311] [info] Loading /usr/share/sdrpp/colormaps/classic.json
[2021-04-24 09:12:04.328] [info] Loading /usr/share/sdrpp/colormaps/greyscale.json
[2021-04-24 09:12:04.390] [info] New DSP samplerate: 768000.0
[2021-04-24 09:12:04.392] [info] AirspyHFSourceModule 'AirspyHF+ Source': Menu Select!
[2021-04-24 09:12:04.392] [info] AirspyHFSourceModule 'AirspyHF+ Source': Tune: 0.0!
[2021-04-24 09:12:04.448] [info] AirspyHFSourceModule 'AirspyHF+ Source': Tune: 100000000.0!
[2021-04-24 09:12:04.448] [info] Ready.
```
Then, here is the error produced when connecting to the server:
```shell
[2021-04-24 09:12:59.814] [info] AirspyHFSourceModule 'AirspyHF+ Source': Menu Deselect!
[2021-04-24 09:12:59.814] [info] New DSP samplerate: 2560000.0
[2021-04-24 09:12:59.815] [info] SpyServerSourceModule 'SpyServer Source': Menu Select!
[2021-04-24 09:13:22.919] [info] SpyServerSourceModule 'SpyServer Source': Tune: 692800.0!
Connected[2021-04-24 09:46:01.867] [info] SpyServerSourceModule 'SpyServer Source': Start!
[2021-04-24 09:46:01.867] [info] SpyServerSourceModule 'SpyServer Source': Tune: 692800.0!
[2021-04-24 09:46:01.867] [error] Socket error
```

AlexandreRouma, 2021-04-24:
Hello,
The spyserver source is currently non-functional.

Best regards,
Alex

10man, 2021-04-24:
Maybe I can help, what is missing? Is there a new protocol that need implemented or are the changes going to be deeper than that?

AlexandreRouma, 2021-04-24:
Most of it actually. The spyserver protocol is really badly "documented" and the module was just a quick test to see if it was even possible to interface with it. It needs a full rewrite and some cleaner cross platform networking code.

10man, 2021-04-24:
I can't even find the source repo for the spyserver itself, much less any API docs? Do you have a place to start looking for the code or the API?

AlexandreRouma, 2021-04-24:
> I can't even find the source repo for the spyserver itself

Both spyserver and sdr# are proprietary so that's normal. You can find the "docs" (which are actually just raw C# code...) on the airspy groups.io, I don't have them with me atm.

nmaster2042, 2021-04-24:
Here is the raw C# code
[SpyServerClient.zip](https://github.com/AlexandreRouma/SDRPlusPlus/files/6370895/SpyServerClient.zip)

AlexandreRouma, 2021-04-24:
last time I tried I wasn't able to get that client to work, i'll try again I guess

10man, 2021-04-25:
I wonder if it's not just simpler to do a server mode of sdr++ that provides an open source protocol / interface that allows remote connection and control of the receiver. Being that spyserver is proprietary and the protocol to the receiver isn't documented, why not just go around it.

It really comes down to how tough it would be to run the airspyhf_source as a server. Could we UDP the data on a port and setup a tcp port for control / feedback.

Thoughts?

nmaster2042, 2021-04-25:
@10man Yes it could be a real alternative to make a sdrpp_server that can use sdrpp existing sources, so all sdr hardware supported by sdrpp would be remotely usable. SDR console have this feature.

AlexandreRouma, 2021-04-25:
it is already planned but I don't want to rush it. It won't be anything like spyserver or rtl_tcp, much more advanced

AlexandreRouma, 2021-07-26:
the spyserver source is now functional

closed 2021-07-26 by AlexandreRouma