AlexandreRouma/SDRPlusPlus #110: Problems with precompiled deb-packages - Debian 10/11

DL5SMB, 2021-04-25:
Hi Alexander,
as already written on Twitter, I tried the new precompiled Deb-packages with my Debian 10 machine.

It did not start as it expected libvolk2-dev and Debian 10 provides only libvolk1-dev.
So  -as suggested - I linked the the missing libvolk.so.2.2  to the old libvolk.so.1.4.
Then it started, but another problem with glibc appeared.

So I upgraded everything to Debian 11/ Bullseye and tried again.

This time a link from the existing libvolk.so.2.4 to the demanded libvolk.so.2.2 was necessary
to make it start.

But after starting it crashed again with the following error message:

`[2021-04-25 09:58:19.934] [info] Initializing Recorder (recorder)
terminate called after throwing an instance of 'nlohmann::detail::type_error'
  what():  [json.exception.type_error.302] type must be number, but is null
Abgebrochen
`
Sascha

AlexandreRouma, 2021-04-25:
There seems to be an issue with the recorder's config getting corrupted in the event of some crashes. Try deleting `~/.config/sdrpp/recorder_config.json`, it'll recreated a clean one.

Regards,
Alex

DL5SMB, 2021-04-25:
Thanks, that did the trick. SDR++ is running now.

I can select and open the plugged in devices (RTL-SDR and Pluto), but they do not work on Debian 11 any more.
The old preview worked on Debian 10, but either something went wrong during upgrading or theree is a bug in Debian 11.
I can open the devices and get something out of the speakers, but  I cant receive a real signal.
On the Pluto there is not even a change in the spectrum while changing frequencies. And the rtl-sdr-blog v3 seems to have some problems with tuning (no PLL locking) and it even crashes, if I try to make bigger frequency changes.

`Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 1792000.030382 Hz
[R82XX] PLL not locked!
[R82XX] PLL not locked!
[INFO] Using format CF32.
[2021-04-25 19:37:32.206] [info] SoapyModule 'SoapySDR Source': Start!
Allocating 15 zero-copy buffers
[R82XX] PLL not locked!
[2021-04-25 19:37:32.261] [info] SoapyModule 'SoapySDR Source': Tune: 100501962440.33594!
[R82XX] PLL not locked!
[2021-04-25 19:37:37.058] [info] SoapyModule 'SoapySDR Source': Tune: 100401962440.33594!
[R82XX] PLL not locked!
[2021-04-25 19:38:44.518] [info] SoapyModule 'SoapySDR Source': Tune: 90401476420.33594!
[R82XX] No valid PLL values for 3093815092 Hz!
r82xx_set_freq: failed=-1
[2021-04-25 19:38:46.278] [info] SoapyModule 'SoapySDR Source': Tune: 80401476420.33594!
Gleitkomma-Ausnahme
`

DL5SMB, 2021-04-26:
I just updated to the newest precompiled package and deinstalled soapy. Now it works as it is supposed to.
Thanks a lot!

closed 2021-04-26 by DL5SMB