AlexandreRouma/SDRPlusPlus #115: [BUG] macOS Catalina compilation issue: no template named 'EventHandler'

atlex, 2021-04-26:
Mac OS install with MacPorts

Install https://www.macports.org

```
$ sudo port selfupdate
$ sudo port upgrade outdated
$ sudo port install cmake fftw3 glfw glew libvolk SoapySDR airspyhf airspy libiio libad9361-iio hackrf rtl-sdr rtaudio
$ git clone https://github.com/AlexandreRouma/SDRPlusPlus.git
$ cd SDRPlusPlus && mkdir build && cd build && cmake .. && sudo make -j4 && sudo make install
```

In file included from /Users/alex/Documents/SDRPlusPlus/core/src/signal_path/signal_path.h:5:
/Users/alex/Documents/SDRPlusPlus/core/src/signal_path/sink.h:28:54: **error: no template named 'EventHandler'**
        Stream(dsp::stream<dsp::stereo_t>* in, const EventHandler<float>& srChangeHandler, float sampleRate);
                                                     ^
/Users/alex/Documents/SDRPlusPlus/core/src/signal_path/sink.h:30:57: **error: no template named 'EventHandler'**
        void init(dsp::stream<dsp::stereo_t>* in, const EventHandler<float>& srChangeHandler, float sampleRate);
                                                        ^
/Users/alex/Documents/SDRPlusPlus/core/src/signal_path/sink.h:51:9: **error: no template named 'Event'**
        Event<float> srChange;        
        ^
2 warnings and 3 errors generated.
make[2]: *** [soapy_source/CMakeFiles/soapy_source.dir/src/main.cpp.o] Error 1
make[1]: *** [soapy_source/CMakeFiles/soapy_source.dir/all] Error 2
make: *** [all] Error 2

Full log:

[SDRPlusPlus-make-log.txt](https://github.com/AlexandreRouma/SDRPlusPlus/files/6374845/SDRPlusPlus-make-log.txt)

jordancrombie, 2021-04-26:
I can add (as I looked at this for quite a while last), the issue seems to center around the last fix that was added for OSX (just look for event handler in the code).  My guess is it's a small mac specific compile wrinkle...as the event handler appears not to be available when compiling (despite working everywhere else).

Also interesting is the fact that it will only fail to compile if you add a local SDR source.  If you compile with the TCP remote, the code will finish clean.  Which makes sense as it looks like it's blowing up in the sample transform area.

AlexandreRouma, 2021-05-09:
Bug fixed in latest commit

closed 2021-05-09 by AlexandreRouma