AlexandreRouma/SDRPlusPlus #117: JSON Files Should Be Dynamically Loaded When Application Starts

JoshuaKimsey, 2021-04-27:
I discovered this issue when I was working on adding a custom band plan for the USA. I had to `sudo make install` each time I added or changed a JSON file to make it appear in the menus. Preferably, the program would look each time it is started to see if a new JSON file has been added to the right folder, and load it dynamically. Granted, for a true release, this would probably require a dedicated directory in the user's home or Documents, but that's not a bad thing overall. Especially when custom plugins are also gonna be on the table one day. Could even have the custom folder be changeable in a preferences or config file. Or do like some programs do and just have a set number of places it looks, such as: .sdrpp and /home/user/sdrpp.

AlexandreRouma, 2021-04-27:
JSON resource files are stored in /usr/share/sdrpp/ and config json files are stored in /home/users/.config/sdrpp/

JoshuaKimsey, 2021-04-28:
Ah ok! I wasn't aware that separate config JSON files were already stored in a .config folder! Thank you!

closed 2021-04-28 by JoshuaKimsey