AlexandreRouma/SDRPlusPlus #119: Further Compile Issues on OSX - 11.2.3 (Big Sur)

jordancrombie, 2021-04-28:
Following on from #115 , it seems like there may be some additional challenges on OSX.

While previously I was able to compile up SDR++ so long as the Individual radio Options were commented out, the latest pulls now are failing to complete the build at all, dying specifically within the linker.

....
....
....
[ 95%] Linking CXX shared library libsdrpp_core.dylib
ld: library not found for -lGLEW
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [core/libsdrpp_core.dylib] Error 1
make[1]: *** [core/CMakeFiles/sdrpp_core.dir/all] Error 2  

OSX Info
Darwin macbook34.local 20.3.0 Darwin Kernel Version 20.3.0: Thu Jan 21 00:07:06 PST 2021; root:xnu-7195.81.3~1/RELEASE_X86_64 x86_64

Using Homebrew for required libs

atlex, 2021-04-28:
I have the same issue on Mac OS Catalina 10.15.7. But glew is here:

$ pkg-config --list-all | grep -I glew
`glew                                glew - The OpenGL Extension Wrangler library`

$ port search glew
```
glew @2.2.0 (graphics, devel)
    OpenGL Extension Wrangler Library

py-glewpy @0.7.4_12 (python, graphics)
    Python bindings for GLEW (OpenGL Extension Wrangler)

py27-glewpy @0.7.4_12 (python, graphics)
    Python bindings for GLEW (OpenGL Extension Wrangler)

Found 3 ports.
```

loewal, 2021-04-28:
I have these installed and get no errors:

/Users/Loewal/Downloads/ **brew list glew**
/usr/local/Cellar/glew/2.2.0_1/bin/glewinfo
/usr/local/Cellar/glew/2.2.0_1/bin/visualinfo
/usr/local/Cellar/glew/2.2.0_1/include/GL/ (3 files)
/usr/local/Cellar/glew/2.2.0_1/lib/libGLEW.2.2.0.dylib
/usr/local/Cellar/glew/2.2.0_1/lib/cmake/ (4 files)
/usr/local/Cellar/glew/2.2.0_1/lib/pkgconfig/glew.pc
/usr/local/Cellar/glew/2.2.0_1/lib/ (3 other files)
/usr/local/Cellar/glew/2.2.0_1/share/doc/ (20 files)

loewal, 2021-04-28:
So I used Homebrew instead of Macports

atlex, 2021-04-28:
I also installed the GLEW with Homebrew.

$ brew list glew
/usr/local/Cellar/glew/2.2.0_1/bin/glewinfo
/usr/local/Cellar/glew/2.2.0_1/bin/visualinfo
/usr/local/Cellar/glew/2.2.0_1/include/GL/ (3 files)
/usr/local/Cellar/glew/2.2.0_1/lib/libGLEW.2.2.0.dylib
/usr/local/Cellar/glew/2.2.0_1/lib/cmake/ (4 files)
/usr/local/Cellar/glew/2.2.0_1/lib/pkgconfig/glew.pc
/usr/local/Cellar/glew/2.2.0_1/lib/ (3 other files)
/usr/local/Cellar/glew/2.2.0_1/share/doc/ (20 files)

and I still have "ld: library not found for -lGLEW".

@loewal you can try git pull and make again.

jordancrombie, 2021-04-29:
Just for completeness (as noted above, I'm using Homebrew)

$ brew list glew
/usr/local/Cellar/glew/2.2.0_1/bin/glewinfo
/usr/local/Cellar/glew/2.2.0_1/bin/visualinfo
/usr/local/Cellar/glew/2.2.0_1/include/GL/ (3 files)
/usr/local/Cellar/glew/2.2.0_1/lib/libGLEW.2.2.0.dylib
/usr/local/Cellar/glew/2.2.0_1/lib/cmake/ (4 files)
/usr/local/Cellar/glew/2.2.0_1/lib/pkgconfig/glew.pc
/usr/local/Cellar/glew/2.2.0_1/lib/ (3 other files)
/usr/local/Cellar/glew/2.2.0_1/share/doc/ (20 files)

However I did notice some relative wonkiness within the Cellar for Homebrew Glew...(quite a bit of version Sym-linking).

lrwxr-xr-x   1 jcrombie  admin      17 Mar 15  2020 libGLEW.dylib -> libGLEW.2.2.dylib
-r--r--r--   1 jcrombie  admin  660880 Mar 15  2020 libGLEW.a
lrwxr-xr-x   1 jcrombie  admin      19 Mar 15  2020 libGLEW.2.2.dylib -> libGLEW.2.2.0.dylib
drwxr-xr-x   3 jcrombie  admin      96 Mar 15  2020 cmake
drwxr-xr-x   8 jcrombie  admin     256 Mar 15  2020 .
-r--r--r--   1 jcrombie  admin  518336 Apr 26 23:52 libGLEW.2.2.0.dylib
drwxr-xr-x   3 jcrombie  admin      96 Apr 26 23:52 pkgconfig
drwxr-xr-x  10 jcrombie  admin     320 Apr 26 23:52 ..

$ md5 /usr/local/Cellar/glew/2.2.0_1/lib/libGLEW.2.2.0.dylib
MD5 (/usr/local/Cellar/glew/2.2.0_1/lib/libGLEW.2.2.0.dylib) = 01cdf06b729b494b69f2d3e61a47f75d

$ md5 /usr/local/lib/libGLEW.*
MD5 (/usr/local/lib/libGLEW.2.2.0.dylib) = 01cdf06b729b494b69f2d3e61a47f75d (Sym-link)
MD5 (/usr/local/lib/libGLEW.2.2.dylib) = 01cdf06b729b494b69f2d3e61a47f75d  (Sym-link)
MD5 (/usr/local/lib/libGLEW.a) = 702c06ef9c8181b173896e403f89638f
MD5 (/usr/local/lib/libGLEW.dylib) = 01cdf06b729b494b69f2d3e61a47f75d (Sym-link)

Im going to rip it all out and try again....

AlexandreRouma, 2021-04-29:
Hello,

could you run `pkg-config --libs glew` and show what it returns?

jordancrombie, 2021-04-29:
I just ripped it all out, and reinstalled (so I dont have the command run before that)
However I did run just before I did:
$ pkg-config --list-all | grep -I glew
glew                                glew - The OpenGL Extension Wrangler library

That said, even after reinstalling, and cleaning up brew, still get the error:
[ 63%] Linking CXX shared library libsdrpp_core.dylib

ld: library not found for -lGLEW
clang: error: linker command failed with exit code 1 (use -v to see invocation)

To that end, here is the command output as requested:
macbook34:build jcrombie$ pkg-config --libs glew
-L/usr/local/Cellar/glew/2.2.0_1/lib -lGLEW

AlexandreRouma, 2021-04-29:
try `ls /usr/local/Cellar/glew/2.2.0_1/lib ` and see which files are in there
for some reason I remember it being libglew.dylib and not libGLEW.dylib.

jordancrombie, 2021-04-29:
I think we are thinking along the same lines:

macbook34:build jcrombie$ ls /usr/local/Cellar/glew/2.2.0_1/lib
cmake			libGLEW.2.2.0.dylib	libGLEW.2.2.dylib	libGLEW.a		libGLEW.dylib		pkgconfig

On a hunch I tired adding first glew, then fftw, etc) in the direct paths to LIBRARY_PATHS (which of course breaks other things, needing to one by one add them in), however I WAS able to get past the compile error (it finds the libs).  Of course because the LIBRARY_PATHS is hand spun, it cant find anything (its now dying looking for Soapy).. 

macbook34:build jcrombie$ export LIBRARY_PATH="$LIBRARY_PATH:/usr/local/Cellar/glew/2.2.0_1/lib:/usr/local/Cellar/fftw/3.3.9/lib:/usr/local/Cellar/glfw/3.3.4/lib:/usr/local/Cellar/volk/2.4.1_1/lib"
macbook34:build jcrombie$ make
[  1%] Linking CXX shared library libsdrpp_core.dylib
[ 56%] Built target sdrpp_core
[ 57%] Building CXX object CMakeFiles/sdrpp.dir/src/main.cpp.o
[ 59%] Linking CXX executable sdrpp
[ 59%] Built target sdrpp
[ 59%] Built target do_always
[ 60%] Building CXX object radio/CMakeFiles/radio.dir/src/main.cpp.o

jordancrombie, 2021-04-29:
Got it to fully compile up by doing the following (start building as normal):
.....
[ 85%] Linking CXX shared library libsdrpp_core.dylib
ld: library not found for -lGLEW
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [core/libsdrpp_core.dylib] Error 1
make[1]: *** [core/CMakeFiles/sdrpp_core.dir/all] Error 2
make: *** [all] Error 2

Then export the libs into place for the linker....  For some reason only adding glew, causes it not be able to find the rest of the libs..

$export LIBRARY_PATH="$LIBRARY_PATH:/usr/local/Cellar/glew/2.2.0_1/lib:/usr/local/Cellar/fftw/3.3.9/lib:/usr/local/Cellar/glfw/3.3.4/lib:/usr/local/Cellar/volk/2.4.1_1/lib:/usr/local/Cellar/soapysdr/0.8.0/lib:/usr/local/Cellar/airspyhf/1.0/include"
macbook34:build jcrombie$ make -j8
Consolidate compiler generated dependencies of target sdrpp_core
[  1%] Linking CXX shared library libsdrpp_core.dylib
[ 85%] Built target sdrpp_core
[ 92%] Building CXX object file_source/CMakeFiles/file_source.dir/src/main.cpp.o
[ 92%] Building CXX object radio/CMakeFiles/radio.dir/src/main.cpp.o
[ 92%] Building CXX object recorder/CMakeFiles/recorder.dir/src/main.cpp.o
[ 92%] Building CXX object CMakeFiles/sdrpp.dir/src/main.cpp.o
[ 94%] Linking CXX executable sdrpp
[ 94%] Built target sdrpp
[ 94%] Built target do_always
5 warnings generated.
[ 96%] Linking CXX shared library radio.dylib
[ 96%] Built target radio
2 warnings generated.
[ 98%] Linking CXX shared library file_source.dylib
[ 98%] Built target file_source
2 warnings generated.
[100%] Linking CXX shared library recorder.dylib
[100%] Built target recorder

Its something subtle...

atlex, 2021-04-29:
No luck for me. Still have "ld: library not found for -lGLEW".

AlexandreRouma, 2021-04-29:
I think I know what the issue is, a few days ago I removed "target_link_directoriy"  cmake commands to allow compiling on older cmake versions. I'll try adding them back and see if it fixes it.

loewal, 2021-04-29:
I had to remove the references to airspy, airspyhf, perseus and bladerf in the main makefile. Then it compiles, otherwise things like airspyhf.h etc. are not found. I pulled the airspyhf from airspy and copied the necessary files, but later on things like lbiio prevent it to continue. 
(#include <libairspyhf/airspyhf.h> in src-directory.....where is this interms of real location?)

In Linux all went fine

loewal, 2021-04-29:
Pulled most recent source:
/Users/Loewal/Downloads/SDRPlusPlus/airspyhf_source/src/main.cpp:10:10: fatal error: 'libairspyhf/airspyhf.h' file not found
#include <libairspyhf/airspyhf.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~
2 warnings and 1 error generated.
make[2]: *** [airspyhf_source/CMakeFiles/airspyhf_source.dir/src/main.cpp.o] Error 1
make[1]: *** [airspyhf_source/CMakeFiles/airspyhf_source.dir/all] Error 2
make: *** [all] Error 2

loewal, 2021-04-29:
This was after missing sources were edited.

loewal, 2021-04-29:
Pulled again after the airspy-change in the source:

Same error as above...or is it an ongoing project and should we wait?

AlexandreRouma, 2021-04-29:
The airspy include stuff should be fixed now, but i'm still running into the weird EventManager thingy :/

jordancrombie, 2021-04-30:
Can confirm these changes seem to have done the trick.  Still seeing the #115 issue as well, but I think this one looks fixed. Maybe @loewal wants to chime in?  If not Ill close this.

loewal, 2021-04-30:
Tried again this morning.

#include <libairspyhf/airspyhf.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~
2 warnings and 1 error generated.
make[2]: *** [airspyhf_source/CMakeFiles/airspyhf_source.dir/src/main.cpp.o] Error 1
make[1]: *** [airspyhf_source/CMakeFiles/airspyhf_source.dir/all] Error 2
make: *** [all] Error 2

Is there something I forget Alexander?

loewal, 2021-04-30:
I removed references to airspy, airspyhf, bladerf and forgot perseus.

This happened:

#include <iio.h>
         ^~~~~~~
2 warnings and 1 error generated.
make[2]: *** [plutosdr_source/CMakeFiles/plutosdr_source.dir/src/main.cpp.o] Error 1
make[1]: *** [plutosdr_source/CMakeFiles/plutosdr_source.dir/all] Error 2
make: *** [all] Error 2

Any help?

loewal, 2021-04-30:
Removed also the perseus-reference in the main cmake-file.
Compiles well.
sdrpp ----> SoapySDR ----> segmentation fault
sdrpp ----> Hackrf ----> crash

AlexandreRouma, 2021-04-30:
Yje hackrf module can't be used along with the soapysdr module. You need to delete one of them.

AlexandreRouma, 2021-05-03:
Just found the issue for the EventHandler thingy. Will push a fix shortly.

AlexandreRouma, 2021-05-09:
closing since MacOS compile issues seem to have been fixed.

closed 2021-05-09 by AlexandreRouma

loewal, 2021-05-10:
Still having issues....

/Users/Loewal/Downloads/SDRPlusPlus/plutosdr_source/src/main.cpp:8:10: fatal error: 'iio.h' file not found
#include <iio.h>
         ^~~~~~~
2 warnings and 1 error generated.
make[2]: *** [plutosdr_source/CMakeFiles/plutosdr_source.dir/src/main.cpp.o] Error 1
make[1]: *** [plutosdr_source/CMakeFiles/plutosdr_source.dir/all] Error 2
make: *** [all] Error 2

How to install libiio?

loewal, 2021-05-10:
I did:
brew install libiio but still not found...

loewal, 2021-05-10:
I am also working with SDRangel and there the compilation of Pluto goes well. Also libiio....