AlexandreRouma/SDRPlusPlus #121: Should new additions/modules by outside contributors require commented code?

JoshuaKimsey, 2021-04-28:
As time moves along and more people wish to contribute external modules to the program, I was wondering if some kind of code commenting should be required? Doesn't have to be overly deep or in-depth commenting, not trying to impress any academics with it... just simple comments for major code blocks to make it easier to identify and understand what the code author is trying to achieve with their modules code. The main reason I might suggest this is for security's sake, but also for clarity to both the creator and contributors of SDR++, as well as other outside contributors too. 

If you like this idea, I can start working up a basic commenting plan for SDR++ and how to best describe it in the Contributing.md file. Let me know what you think!

AlexandreRouma, 2021-04-28:
Proper comments of all non-trivial parts of submitted code are required.
I'll add proper guidelines this weekend to further describe the code and comment style.

closed 2021-04-28 by AlexandreRouma

JoshuaKimsey, 2021-04-28:
Cool! Yeah, the other reason I asked is because here shortly I'm gonna see if I can start work on a module that can decode NOAA APT directly from SDR++, either on its own or by integrating with another program like WXtoIMG or NoaaAPT. Perhaps have both options be available so users can either have a simplified decoding directly from SDR++ or have more feature rich options with the other two. In doing this, I wanted to make sure I follow proper commenting rules for my code.

AlexandreRouma, 2021-04-29:
I'm actually already working on such a module. The universal weather satellite decoder module will container live decoders for:
- NOAA DSB
- NOAA HRPT
- NOAA APT
- Meteor LRPT
- Meteor HRPT
- Fengyun AHRPT
- Metop AHRPT

The code I've pushed to the repo only contains the NOAA hrpt decoder but my personal satellite decoder (which isn't public) already contains NOAA DSB+APT+HRPT and Meteor LRPT+HRPT

JoshuaKimsey, 2021-04-29:
Ok, that's awesome! And also makes me kinda glad, cause I wasn't quite sure where to begin with trying to make my own module... >.>

I will say though, I don't know how feature rich your APT decoder is/will be, but you _could_ consider integrating features from the open source [AptDec](https://github.com/Xerbo/aptdec), to add features such as coloration and channel control. It's also written in C, so integration shouldn't be too hard. 

I can't wait to try out your weather satellite decoder though! You'll have to give some info on how to use such features when you launch them with SDR++. As a Linux user, I certainly thank you for creating a more unified experience! One that doesn't require a minimum of 4 steps to acquire an image: Orbitron (Satellite Tracking) >GQRX (Audio Recording) >NoaaAPT (Check to make sure image was recorded + resampling) >WXtoIMG (Image coloration and processing). Even just being able to integrate 2 or 3 of these steps would be amazing!

AlexandreRouma, 2021-04-29:
> It's also written in C, so integration shouldn't be too hard.

tbh i'd rather not copy any code. SDR++ is a learning project for me, I wrote the entire DSP and UI from scratch for a reason.
I'll try to add the same *features* but I don't want to take their code and put it in SDR++ with few modifications.