AlexandreRouma/SDRPlusPlus #123: Recorded file paths & SDRplay native interface

Simon-Atkinson, 2021-04-29:
Hi Alex,
On the Linux version, the path for recorded files is not searchable. It has to be entered manually, whereas on the Windows version, it does appear to be searchable.
[image]

One other point, decimation with appropriate filtering is available within the API as is support for Low IF mode (fully converted to I/Q baseband at the API output). You just need to make the appropriate calls to the API. Are these features that you feel you might be able to add for the native support of SDRplay devices?

AlexandreRouma, 2021-04-29:
Hello,

Yes, there currently isn't a way to select using the file manager. This is because all linux distros have a different API for it so I haven't implemented a way to do it yes. I'll likely look into some abstraction utils for that soon.

For decimation, I haven't added that yet because I'm working on software decimation inside SDR++, thus I thought using the API for decimation would be redundant (correct me on that if it actually has some benefits I don't know about)

I will add the low IF mode soon. To be honest, I didn't quite understand what the setting does so I didn't try to implement it yet.

Best regards,
Alex

Simon-Atkinson, 2021-04-29:
Hi Alex,

Low IF is a hardware mode in the RSP platform. In fact the RSPduo only uses Low IF mode when operating both tuners simultaneously. It has a fixed sample rate and a fixed 2 MHz bandwidth and under certain circumstances, it can have some advantages. The API automatically de-rotates the IF to I/Q baseband.  It is by no means essential as you have automatic offsetting of the VFO and LO which is important in ZIF mode.

As for decimation, I see your point. There is no value in using the in-built decimation functions with you were going to add it into the core of program for other hardware platforms.

 

Thanks for the explanation regarding the issue with the file manager in Linux.

 

Best regards

 

Simon

AlexandreRouma, 2021-04-29:
I see thanks for clearing this up.
I'll try to add that feature this weekend.

Regards,
Alex

shuyuan-liu, 2021-04-29:
@AlexandreRouma For graphically choosing folders on Linux, [portable-file-dialogs](https://github.com/samhocevar/portable-file-dialogs) is a possible solution. It's a single header file, compatible with Win, Linux and Mac, and uses their respective native file choosers. Aang23's SatDump uses this library and I can confirm it works for me with GNOME and KDE on Linux and on Windows.

AlexandreRouma, 2021-04-29:
Thanks for pointing me to it, that looks like the perfect solution indeed!

AlexandreRouma, 2021-04-29:
portable-file-dialogs has been implemented. You should now be able to select files on linux!

Simon-Atkinson, 2021-04-30:
Works well Alex.

 

Many thanks

 

Simon

AlexandreRouma, 2021-07-26:
Decimation has been added through the software itself.

Will add the low IF mode tot he source module soon.

AlexandreRouma, 2022-07-16:
this was implemented a while back

closed 2022-07-16 by AlexandreRouma