AlexandreRouma/SDRPlusPlus #124: Show plug-in name with each demodulator on spectrum

shuyuan-liu, 2021-04-29:
When trying out the Meteor demodulator plugin, I found it slightly difficult to distinguish between the main demodulator frequency (the yellow vertical line in the screenshot below) and that of the Meteor demodulator (the red line) on the spectrum. When moving them around it is easy for the user to lose track, and they change colours between red and yellow depending on which one was moved last, so that doesn't tell you which is which either. I think it would be clearer if each demodulator (i.e. "highlight" if that makes sense) shows the name of the plugin it belongs to, for example "Radio" and "Meteor Demodulator".

P.S. What's the correct term to refer to the "highlight" or "currently tuned frequency"? I see many people using VFO but that seems to stand for variable frequency oscillator, which imo isn't really accurate.

[image]

AlexandreRouma, 2021-04-29:
Hello,
The VFO highlighted red is the currently selected VFO. A way of coloring VFOs to distinguish them easily is planned (as well as showing the name when hovering over for a second)

And yes, VFO is the right term. It's a reference to old super-heterodyne radios that used a variable frequency oscillator to tune around.

Regards,
Alex

AlexandreRouma, 2021-05-02:
VFO names are now showed when hovering.

shuyuan-liu, 2021-05-08:
Thanks, that's great! VFO colouring has been implemented too.